thonny/thonny #770: step over steps over a loop

michAtEl, 2019-06-02:
The Debugger-Command "Step over" (f6) steps over a loop. I would prefer a different behaviour. In my opinion, "Step over" should go into the loop.

There are two reasons why I prefer this:
- all debuggers I know step into a loop when i use "Step over"
- I think this bevaviour would be easier for beginners because they can use on Command ("Step over") to go through the instructions fo their program. At the moment, they need "Step into" if the next instruction is a loop and "Stop over" if the next instruction is not a loop.

aivarannamaa, 2019-06-03:
Thank you for the suggestion! This is an important topic for me and I'm eager to explain my viewpoint.

Short story: This debugger (the "nicer" one) is primarily meant to highlight and explain the program structure, including compound statements like `if` and `while`. It's not meant to be comfortable for people who already understand Python or some other imperative programming language (but I think it can be comfortable, if you get more familiar with it).

> all debuggers I know step into a loop when i use "Step over"

Do you know why the other debuggers step **into** a loop when you use "Step **over**"? Because they usually only have line numbers available at debug time -- all other information about original program structure is lost. Furthermore, I believe most people in Thonny's target group are not familiar with any debuggers, so IMO the familiarity argument is not relevant here.

> I think this behaviour would be easier for beginners because they can use on Command ("Step over") to go through the instructions of their program. At the moment, they need "Step into" if the next instruction is a loop and "Step over" if the next instruction is not a loop.

But what if they want to investigate a piece of code which is just after a 100-iteration loop? Traditional debuggers do allow user to choose the desired granularity of stepping ("into" vs "over"), but for technical reasons they allow this choice only at function calls. Thonny's "nicer" debugger has stepped over these difficulties (by sacrificing performance) and is able to offer this choice at all levels.

More importantly, during the years I've taught programming, I've realized that easy is not always effective. I do want my students to spend time on thinking whether they want to use "step into" or "step over" at certain point in their code. I find it very important that they understand the structure of Python programs as early as possible and the "nicer" debugger has been designed to support this understanding. [This video](https://www.youtube.com/watch?v=nwIgxrXP-X4) hopefully gives a hint about how I've used Thonny to explain Python language it in my lessons.

As I said above, the "nicer" debugger has sacrified performance in order to be more knowledgeable about program structure. Occasionally it's overhead may become prohibitive. Therefore I created an alternative, a "faster" debugger (Shift+F5), which works pretty much like traditional debuggers. You may want to use this if you really fail to make friends with the "nicer" one :)

Also, don't forget that you can turn off each-stack-frame-in-separate-window visualization in options, if you prefer more traditional style (https://github.com/thonny/thonny/blob/master/thonny/plugins/help/debuggers.rst)

michAtEl, 2019-06-04:
Thank you for your explanation!
I was not aware that there are multiple debuggers that behave differently (I'm still new to Python and Thonny).

closed 2019-06-04 by michAtEl