thonny/thonny #778: Run code without a priori saving

gvitalie, 2019-06-12:
Could we please have somehow such feature:

_

> run code without a priori saving to file

_

Thank You, Dear Thonny.

juanfal, 2019-06-12:
I agree with @gvitalie, it's quite frequent you, in class, jot down a simple piece of program and try to execute it.  It doesn't need to have a purpose and less a name and file.
Thanks @aivarannamaa

CountryTk, 2019-06-13:
@juanfal Running a file would mean you'd still need to save the file and have a name for it but that could be done automatically like generating a random name for it

aivarannamaa, 2019-06-13:
Actually Thonny doesn't need a file in order to run editor content. I think I'll make it so that untitled editors won't need to be saved for running, but once the editor gets a name, it will be still automatically saved before each run.

CountryTk, 2019-06-13:
@aivarannamaa Do you have a public email address I could contact you with? I have some questions about Thonny!

aivarannamaa, 2019-06-14:
> Do you have a public email address I could contact you with? I have some questions about Thonny!

Sure, combine my names with a period and append Google's e-mail domain.

juanfal, 2019-06-14:
The only problem IMHO I see with this, is the current working directory could be funny, especially if you are treating files (opening, using external files).  But for that, python can always be called with -c sending an on the fly text with the very python source code.

Only a suggestion.  It can be quite useful for quick and dirty very short programs you can get rid of easily and at the same time showing how easy is writing them.  A dynamic way to encourage them to start trying.  

- Juan F.

aivarannamaa, 2019-06-14:
> The only problem IMHO I see with this, is the current working directory could be funny, especially if you are treating files (opening, using external files). 

Fortunately Thonny 2.2 will allow selecting working directory via Files view and nameless runs would use that directory.

aivarannamaa, 2019-06-14:
Related to #318 and https://bugs.python.org/issue19042

The-Bow-Hunter, 2021-06-24:
jsut asking but why dont you just send it via REPL just as the console?

aivarannamaa, 2021-06-25:
> jsut asking but why dont you just send it via REPL just as the console?

This is more or less what I plan to do. I just haven't got to it yet. Beside minimal solution there are several details to consider (finding the correct editor when clicking the stactrace, Pylint results etc)

closed 2022-02-01 by aivarannamaa

jouellnyc, 2022-06-20:
New Thonny user here. Google'd to get here. This is running great in 4.0.0 b3. It is very intuitively implemented and as a new user it's really helpful.Thanks!