thonny/thonny #780: Prepare main code for Python 3.8

aivarannamaa, 2019-06-14:


aivarannamaa, 2019-06-14:
cee7a79

Ivoz, 2019-07-19:
If 3.8 is coming out October-ish, do you want to consider making 3.2 able to handle it already?

aivarannamaa, 2019-07-19:
Actually, I think it does handle it already -- at least the debugger won't choke on walrus operator and new features in f-strings.

I'll do more thorough testing when first RC of 3.8 comes out.

Ivoz, 2019-07-19:
Do you know what you thought would be incompatibilities?

aivarannamaa, 2019-07-19:
Initially some of the tests related to AST analysis failed. I solved this by deleting the tests :)

I'm now using [asttokens](https://github.com/gristlabs/asttokens) for this and they are already testing same things, so I don't have to. It looks like they are now ready for Python 3.8.

closed 2019-08-24 by aivarannamaa