thonny/thonny #787: Frequent fatal error on reconnect

dadown0, 2019-06-20:
When I disconnect my ESP32 to make a hardware breadboard change and then reconnect it, I almost always get a fatal error in Thonny when trying to use REPL again. Then after starting Thonny again, It has forgotten the configuration settings and reverted to the internal interpreter.

Here is what I get in the popup error report:
Traceback (most recent call last):
  File "C:\Users\Dave\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\Dave\AppData\Local\Programs\Thonny\lib\site-packages\thonny\tktextext.py", line 233, in dispatch
    return fun(event)
  File "C:\Users\Dave\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 382, in perform_return
    EnhancedTextWithLogging.perform_return(self, event)
  File "C:\Users\Dave\AppData\Local\Programs\Thonny\lib\site-packages\thonny\tktextext.py", line 408, in perform_return
    self.insert("insert", "\n")
  File "C:\Users\Dave\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 343, in intercept_insert
    self._try_submit_input()
  File "C:\Users\Dave\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 493, in _try_submit_input
    self._submit_input(submittable_text)
  File "C:\Users\Dave\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 636, in _submit_input
    get_runner().send_program_input(text_to_be_submitted)
  File "C:\Users\Dave\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 237, in send_program_input
    self._proxy.send_program_input(data)
  File "C:\Users\Dave\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\__init__.py", line 127, in send_program_input
    assert self._serial.outgoing_is_empty()
AttributeError: 'NoneType' object has no attribute 'outgoing_is_empty'

aivarannamaa, 2019-06-21:
Thanks for the report!

I have plans for refactoring the code so that I can find bugs like this more easily

ivancho523, 2019-06-27:
i have the same problem

aivarannamaa, 2019-07-06:
MicroPython back-end was thoroughly refactored for upcoming Thonny 3.2b2. 

Hopefully these changes had fixed your problem (or at least changed the error). If you still have problems with 3.2, then please create another issue.

closed 2019-07-06 by aivarannamaa