thonny/thonny #789: Right-click will undo last action

9code7, 2019-06-22:
Hello,
When I right-click in the editor, it functions as "undo." This happens on the up-click of the right button. I have to type something into the editor before the right-click functions as "undo."

I'm using Thonny 3.1.2 on Ubuntu 16.04.6, Python 3.7.2 (64-bit), Tk 8.6.9, and am using the Clean Dark Blue and Default Dark themes.

aivarannamaa, 2019-06-22:
Thank you for the report! This is duplicate of #731 and will be fixed in version 3.2.

In the meanwhile, the workaround is to use "Enhanced Clam" theme.

closed 2019-06-22 by aivarannamaa

9code7, 2019-06-22:
Thanks for the workaround! 
Confirmed: right-click functions as expected after changing to "Enhanced Clam" and restarting Thonny.