thonny/thonny #793: A lot of Super commands are missing.

jdcrunchman, 2019-07-02:
I'm using Thonny,  but perplexed by a "Show stopper"
First off,  there is no Super command for creating a "www" directory from within Thonny.
In order to do that,  I had to quit Thonny.   Then from my Mac laptop
issue a command:    ampy --port /dev/tty.SLAB_USBtoUART mkdir www
 Then I started Thonny...   and issued the following command to verify the Directory was created

>>> %lsdevice
boot.py
test.py
microWebSrv.py
microWebSocket.py
microWebTemplate.py
mainweb.py
Networking.py
trace.py
startup.py
hotspot.py
www     <---- here it is.

This link --- https://github.com/thonny/thonny/blob/master/thonny/plugins/micropython/__init__.py 
Is the python source code,  where I was able to find descriptions of the commands in line 105
The "ls" command was defined there in the source.

>>> %cd /www
Path doesn't exist: /www

So this is a true "show stopper" with very limited control over the files managed by Thonny.
when will there be an update available to make it easier to use Thonny.

Now I'm seriously planning to go back to Atom IDE and try once again to get that ESP32 Plugin working....  But thats for the Atom forum,  not here.

John

aivarannamaa, 2019-07-06:
I'll probably add `mkdir` command in a future version.

In the meanwhile you can use following:

```
>>> import os
>>> os.mkdir("/www")

```

jdcrunchman, 2019-07-06:
Yea,  this has been a show stopper for me.    But I've already tried that
earlier.    Also,   I have a crash trace for you.
Thonny crashes a lot,   but I can restart it again,  but it's rather
annoying when developing.

Can you use this crash trace to figure out whats going on?

John

aivarannamaa, 2019-07-06:
Did you forget to attach the stacktrace?

jdcrunchman, 2019-07-08:
No,   I didn't - gmail apparently did not allow me to send it as an
attachment.
So I'm sending it in the body of this message..     Do you have a cloud
storage system?
I can send it there if this don't work.

Process:               Python [1383]

Path:
/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python

Identifier:            org.thonny

Version:               0.0.0 (0.0.0)

Code Type:             X86-64 (Native)

Parent Process:        ??? [1]

Responsible:           Python [1383]

User ID:               501

Date/Time:             2019-07-05 09:37:41.614 -0700

OS Version:            Mac OS X 10.13.6 (17G7024)

Report Version:        12

Anonymous UUID:        DC99814D-9225-1879-AD9B-FF1D5AD1F530

Sleep/Wake UUID:       0EA85E8F-064E-45D9-8123-E64968C39153

Time Awake Since Boot: 28000 seconds

Time Since Wake:       630 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)

Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000024

Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x24:

-->

    __TEXT                 000000010a847000-000000010a848000 [    4K]
r-x/rwx SM=COW  s
[/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread

0   libsystem_kernel.dylib        0x00007fff7a0c0b66 __pthread_kill + 10

1   libsystem_pthread.dylib       0x00007fff7a28b080 pthread_kill + 333

2   libsystem_c.dylib             0x00007fff79fce6fe raise + 26

3   libsystem_platform.dylib      0x00007fff7a27ef5a _sigtramp + 26

4   ???                           000000000000000000 0 + 0

5   libtk8.6.dylib                0x000000010b2f04bb CharUndisplayProc + 95

6   libtk8.6.dylib                0x000000010b2e89a1 FreeDLines + 225
[... 1414 lines cut ...]
<351781D5-3CCF-320F-BB5A-4DFCC1CCD31D> /usr/lib/libxar.1.dylib

    0x7fff79c47000 -     0x7fff79d2efff  libxml2.2.dylib (31.13)
<8C12B82A-66FD-330C-9BEA-AAC090C7076A> /usr/lib/libxml2.2.dylib

    0x7fff79d2f000 -     0x7fff79d57fff  libxslt.1.dylib (15.12)
<4A5E011D-8B29-3135-A52B-9A9070ABD752> /usr/lib/libxslt.1.dylib

    0x7fff79d58000 -     0x7fff79d6affb  libz.1.dylib (70)
<48C67CFC-940D-3857-8DAD-857774605352> /usr/lib/libz.1.dylib

    0x7fff79e06000 -     0x7fff79e0aff7  libcache.dylib (80)
<092479CB-1008-3A83-BECF-E115F24D13C1> /usr/lib/system/libcache.dylib

    0x7fff79e0b000 -     0x7fff79e15ff3  libcommonCrypto.dylib (60118.50.1)
<029F5985-9B6E-3DCB-9B96-FD007678C6A7> /usr/lib/system/libcommonCrypto.dylib

    0x7fff79e16000 -     0x7fff79e1dfff  libcompiler_rt.dylib (62)
<968B8E3F-3681-3230-9D78-BB8732024F6E> /usr/lib/system/libcompiler_rt.dylib

    0x7fff79e1e000 -     0x7fff79e27ffb  libcopyfile.dylib (146.50.5)
<3885083D-50D8-3EEC-B481-B2E605180D7F> /usr/lib/system/libcopyfile.dylib

    0x7fff79e28000 -     0x7fff79eadffb  libcorecrypto.dylib (562.70.3)
<37AF8EC6-BF34-3D15-BC3A-A9B98F2427D4> /usr/lib/system/libcorecrypto.d

aivarannamaa, 2019-07-10:
Thank you! Unfortunately these low-level crashes are out of my reach. We can only hope that next version of Tk bundled with Python is more robust than current one.

jdcrunchman, 2019-07-11:
I really like the program,  other then the annoying crash,  it's OK,  but
leaves a lot of things I need which I explained earlier.   Can you please
tell me when to expect another update on the program.

I'm frustrated with Atom (Your competitor) who has this plugin called
PyMakr,  but it only works with PyCom's hardware products,  but when we
want to
roll our own circuit which has far better power management,  the current
PyMakr plugin does not work on our hardware which is very frustrating,  it
can't
seem to find the "device" being an ESX32 micro processor which we are using
with the LoRa SX1276 chip.   I so want that PyMakr work with Atom as it's
my
main IDE.

I've posted this issue on at least two forums (PyCom and Semtech,  but to
no avail.

John

jdcrunchman, 2019-07-12:
I finally got PyMakr working on Atom IDE,  it now connects to the device.
 There was an obscure gotcha that was pointed out
in the PyMakr GitHub bug tracker.

John

aivarannamaa, 2019-07-19:
In case you are still interested, Thonny 3.2.0 beta contains redesigned file handling for MicroPython: https://github.com/thonny/thonny/releases/tag/v3.2.0b7

Some commands (eg. make directory) are still missing, though.

aivarannamaa, 2023-01-26:
Thonny now allows creating directories in MicroPython filesystem.

closed 2023-01-26 by aivarannamaa