thonny/thonny #795: Semi-colon not supported in shell?

peterdcasey, 2019-07-06:
Type: **2 + 3; 4 + 5 |enter|** into the shell and only the last expression evaluated is displayed. Both expressions results should be displayed.

(This is a bit nit-picky, why would anyone really want this?)

Peter

aivarannamaa, 2019-07-07:
Thank you for educating me! I did not know that the official REPL works like this.

I'll try to fix it later.

kajukenbo, 2019-07-29:
For what it is worth, IPython seems to behave the same way.
However "bpython" seems to work as @peterdcasey expects.

At least with my configuration:

 $ ipython
Python 3.7.4 (default, Jul  9 2019, 15:11:16)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.6.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: 2+3;4+5
Out[1]: 9

In [2]:
-------------------------------
 $ bpython
bpython version 0.18 on top of Python 3.7.4 ../bin/python3.7
>>> 2+3;4+5
5
9
>>>

closed 2020-09-01 by aivarannamaa