thonny/thonny #797: Poor fonts installing from pip with conda

kajukenbo, 2019-07-08:
Linux Mint / Ubuntu Xenial LTS, patched
Latest Miniconda; Python 3.7.3
pip install thonny

Fonts and UI are poorly rendered bitmaps.
Truetype fonts are not even listed.

I've re-installed the system from scratch to replicate the issue.
With the way the screenshot is zoomed it, it does not do justice to how bad the fonts really are.
It seems like there may be a missing TK dependency or something but I cannot figure out what it might be.

[image]

kajukenbo, 2019-07-08:
Cinnamon desktop, 64-bit
Intel i915 on-board laptop graphics
2nd generation Intel i3 CPU

aivarannamaa, 2019-07-08:
I have seen it before, but I don't remember which package helped.

Please check the deps for this: https://packages.debian.org/buster/python3-tk

aivarannamaa, 2019-07-08:
I found this: https://stackoverflow.com/questions/47769187/make-anacondas-tkinter-aware-of-system-fonts-or-install-new-fonts-for-anaconda

kajukenbo, 2019-07-10:
Thank you sir.

That looks like the exact issue.  
I have dropped the use of any TK-based apps with conda - that includes the likes of IDLE as well.
For the last couple of days I have been testing successfully with the "deadsnakes" python repos.

For anyone who runs across this, my current solution looks like this:
(2 #s is a comment)

`# sudo add-apt-repository ppa:deadsnakes/ppa --yes
# sudo apt update

## TK build in conda does NOT support truetype
# sudo apt install idle-python3.7 python3.7-venv -y ## pulls python3-tk, ensurepip, etc from deadsnakes-ppa.

# python3.7 -m ensurepip --default-pip
# python3.7 -m pip install --upgrade pip setuptools wheel

## I put a venv "pycode" under ~/Documents with my python code in it
# cd ~/Documents
# python3.7 -m venv pycode  ## once in the venv the default python is 3.7.x
# source pycode/bin/activate
# grep "bin/activate" ~/.bashrc || echo "source ~/Documents/pycode/bin/activate" | tee -a ~/.bashrc

# pip install --upgrade pip  ## in the venv pip defaults to python3.7.x
# pip install thonny idlex ipython
# ipython profile create
`

aivarannamaa, 2019-09-01:
I think there is nothing I can do here.

closed 2019-09-01 by aivarannamaa