thonny/thonny #799: A problem when running script in thonny

dqs8100, 2019-07-08:
I write a class with __del__. Then I run a script in thonny using F5, but when the script ends, the __del__ is not called. I run the same script in terminal, when the script ends, the __del__ is called.
Another problem is I use 'del *' to destory an object in thonny's shell, the __del__ is not called either. But I use 'del *' to destory an object in python shell, the __del__ is called.
So how can I make thonny shell the same as python shell, and make the script run in thonny the same as that runs in terminal.
Thank you very much.

aivarannamaa, 2019-07-08:
Please include the code!

dqs8100, 2019-07-09:
> Please include the code!

`class A:
    def __init__(self):
        print("Init an object from A")
    def __del__(self):
        print("Del an object from A")`
And here is the result:

dqs8100, 2019-07-09:
> Please include the code!

Hello, I have updated the issue with the code included. Thanks.

aivarannamaa, 2019-07-09:
Good question!

Here is the background information: https://stackoverflow.com/a/1481512/261181

After each round of interaction with the interpreter, Thonny will gather the reference for each global and local variables in order to allow heap exploration. That's why Python can't garbage collect you `a`.

dqs8100, 2019-07-09:
> Good question!
>
> Here is the background information: https://stackoverflow.com/a/1481512/261181
>
> After each round of interaction with the interpreter, Thonny will gather the reference for each global [...]

Thank you very much. It is really helpful.
So I think if I have some cleanup script, I'd better write it into a cleanup method and call it when I need.
Or I can use 'del a' in the end of my program to destroy the object.

closed 2019-07-11 by aivarannamaa