thonny/thonny #800: Won't load large python script

Gordon999, 2019-07-08:
My script is about 5900 lines, loads OK with IDLE etc but Thonny won't load it, just hangs.

Script available here as a zip...
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=45246&p=1443213&hilit=astrophotography#p1282884

Gordon999, 2019-07-08:
Just found out it eventually loads BUT takes ages.

Gordon999, 2019-07-08:
and after shutting Thonny down and rebooting my PI4 Thonny now won't run from the menu.

aivarannamaa, 2019-07-10:
Thanks for pushing me! :)

I originally did not care about performance with large files, because Thonny was meant for absolute beginners, who wouldn't write programs larger than 100 lines. Now when it's promoted as the main Python editor on RPi, I need to deal with it.

Probably the main bottleneck is syntax coloring, which is more simplistic than IDLE's. I now found some ways to optimize it (by colorizing incrementally only the visible lines).

I can't promise Thonny will be usable for 5000 line programs on RPi, but you can expect noticeable performance improvements in the next version.

aivarannamaa, 2019-07-11:
I've made good progress -- your script now loads in 7 seconds on my RPi 3B+ and can be edited almost smoothly (except for entering/deleting parens, quotes and `#`, as these invoke heavier coloring and paren-matching processes). You will get better performance if you switch off "Highlight parentheses" from Tools => Options => Editor.

I'm stopping here (for now), because I expect large majority of Thonny users to edit modules well under 1000 lines.

closed 2019-07-11 by aivarannamaa

aivarannamaa, 2019-07-12:
> I'm stopping here [...]

Well, I couldn't :)

Your script now loads in couple of seconds and can be edited smoothly if no breakpoints are set. The only hiccups will be starting and closing multiline strings.

Gordon999, 2019-07-15:
Thanks. Much appreciated.