thonny/thonny #807: Thonny Not working for me

hbot07, 2019-07-13:
Here is the error message i get 
Traceback (most recent call last):
  File "/home/parth/apps/thonny/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/home/parth/apps/thonny/lib/python3.7/site-packages/thonny/running.py", line 317, in _cmd_run_current_script
    self.execute_current("Run")
  File "/home/parth/apps/thonny/lib/python3.7/site-packages/thonny/running.py", line 282, in execute_current
    script_dir = normpath_with_actual_case(os.path.dirname(filename))
  File "/home/parth/apps/thonny/lib/python3.7/posixpath.py", line 156, in dirname
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType

I am new to python

aivarannamaa, 2019-07-13:
Did this error happen when you tried to run a script? Is your script saved to a file?

hbot07, 2019-07-13:
no it is not saved . i have no idea about that . I just wrote the statement
print("Hello World") and pressed run .Thanks !

closed 2019-09-01 by aivarannamaa