thonny/thonny #808: when I use thonny to run multiprosess script,it's doesn't work

Dleta, 2019-07-14:
#the script below 
import multiprocessing as mp
from time import sleep
def ab(x):
    while True:
        sleep(1)
        print(x)
if __name__ == '__main__':
    #mp.set_start_method('spawn')
    #ab('a')
    f1 = mp.Process(target = ab, args=('a'))
    f2 = mp.Process(target = ab, args=('b'))
    f1.start()
    f2.start()
    f1.join()
    f2.join()

aivarannamaa, 2019-07-16:
There are syntax errors in your script...
Or do you get some other errors?

Dleta, 2019-07-17:
the error is so simple,so you can fix i.,after you fixed it,you can find the script run in MS cmd with no problems,but have problems run in thonny and IDLE.
you can try it.

Dleta, 2019-07-17:
I fixed it totally now.

aivarannamaa, 2019-07-17:
I did fix and try it, but didn't notice any problems. Which problems do you mean?

Dleta, 2019-07-17:
??????no problem?why i run it in idle or thonny is no reply?????i run it on windows10 platform,python3.6.8 64bit

Dleta, 2019-07-17:
i should show you a picture,but i do not know how to.

Dleta, 2019-07-17:
the script should print
a
b
a
b
a
.
.
.
but no reply on my computer

aivarannamaa, 2019-07-17:
Hmm, you are right. I works when Thonny is run from my development environment, but stops working when it is installed.

I'll try to find a solution.

Dleta, 2019-07-17:
thanks for you work

aivarannamaa, 2019-07-17:
Thanks for the report!

The fix will be in the next beta of version 3.2.0 (to be published in couple of days)

commit 70ec7594 (2019-07-17): FIXED #808: multiprocessing doesn't work in Windows
closed 2019-07-17 by aivarannamaa