thonny/thonny #813: 3.2.0b4 on OSX fails

juanfal, 2019-07-17:
In the shell it says:

    Backend terminated or disconnected. Use 'Stop/Restart' to restart ...

but pressing the red button Stop/Restart rewrites the same message

No script works

juanfal, 2019-07-17:
I am using OSX beta: 10.15 beta (19A512f)

aivarannamaa, 2019-07-18:
Can you please send me the logs and check whether something interestings appears when it's run from the command line (https://github.com/thonny/thonny/wiki/Troubleshooting) ?

Does Thonny 3.2.1 work properly in the same system?

aivarannamaa, 2019-07-22:
I could not reproduce this in 10.15 beta 19A501i (in VMware)

juanfal, 2019-07-22:
Typing Enter in Shell, it immediately pops up a "Internal Error" window:

    Traceback (most recent call last):
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
        return self.func(*args)
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/tktextext.py", line 229, in dispatch
        return fun(event)
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/shell.py", line 871, in perform_return
        EnhancedTextWithLogging.perform_return(self, event)
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/tktextext.py", line 402, in perform_return
        self.insert("insert", "\n")
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/shell.py", line 833, in intercept_insert
        self._try_submit_input()
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/shell.py", line 1001, in _try_submit_input
        self._submit_input(submittable_text)
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/shell.py", line 1144, in _submit_input
        get_runner().send_program_input(text_to_be_submitted)
      File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/running.py", line 270, in send_program_input
        self._proxy.send_program_input(data)
    AttributeError: 'NoneType' object has no attribute 'send_program_input'

juanfal, 2019-07-22:
I've run the test: 

      python3 /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tkinter/test/runtktests.py | pbcopy 

This is the standard output:

    test_horizontal_range (tkinter.test.test_ttk.test_extensions.LabeledScaleTest) ... ok
    test_initialization (tkinter.test.test_ttk.test_extensions.LabeledScaleTest) ... ok
    test_initialization_no_master (tkinter.test.test_ttk.test_extensions.LabeledScaleTest) ... ok
    test_resize (tkinter.test.test_ttk.test_extensions.LabeledScaleTest) ... ok
    test_variable_change (tkinter.test.test_ttk.test_extensions.LabeledScaleTest) ... ok
    test_widget_destroy (tkinter.test.test_ttk.test_extensions.LabeledScaleTest) ... ok
    test_initialization (tkinter.test.test_ttk.test_extensions.OptionMenuTest) ... ok
    test_menu (tkinter.test.test_ttk.test_extensions.OptionMenuTest) ... ok
    test_unique_radiobuttons (tkinter.test.test_ttk.test_extensions.OptionMenuTest) ... ok
    test_widget_destroy (tkinter.test.test_ttk.test_extensions.OptionMenuTest) ... ok
    patchlevel = 8.6.8
    test_class (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_command (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_compound (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_cursor (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_default (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_image (tkinter.test.test_ttk.test_widgets.ButtonTest) ... skipped 'crashes with Cocoa Tk (issue19733)'
    test_invoke (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_keys (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_padding (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_state (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_style (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_takefocus (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_text (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_textvariable (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_underline (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_width (tkinter.test.test_ttk.test_widgets.ButtonTest) ... ok
    test_class (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_command (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_compound (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_cursor (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_image (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... skipped 'crashes with Cocoa Tk (issue19733)'
    test_invoke (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_keys (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_offvalue (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_onvalue (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_padding (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_state (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_style (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_takefocus (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_text (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_textvariable (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_underline (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_variable (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_width (tkinter.test.test_ttk.test_widgets.CheckbuttonTest) ... ok
    test_background (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_bbox (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_class (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_cursor (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_exportselection (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_font (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_foreground (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_height (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_identify (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... FAIL
    test_invalidcommand (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_justify (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_keys (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_postcommand (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_revalidation (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_show (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_state (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_style (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_takefocus (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
    test_textvariable (tkinter.test.test_ttk.test_widgets.ComboboxTest) ... ok
[... 907 lines cut ...]
    test_get (tkinter.test.test_tkinter.test_variables.TestIntVar) ... ok
    test_invalid_value (tkinter.test.test_tkinter.test_variables.TestIntVar) ... ok
    test_default (tkinter.test.test_tkinter.test_variables.TestDoubleVar) ... ok
    test_get (tkinter.test.test_tkinter.test_variables.TestDoubleVar) ... ok
[... cut ...]
    ----------------------------------------------------------------------
    Traceback (most recent call last):
      File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/tkinter/test/test_ttk/test_widgets.py", line 332, in test_identify
        self.assertEqual(self.entry.identify(5, 5), "textarea")
    AssertionError: 'Combobox.button' != 'textarea'
    - Combobox.button
    + textarea

    ----------------------------------------------------------------------
    Ran 975 tests in 6.873s

    FAILED (failures=1, skipped=17)

juanfal, 2019-07-22:
Any other tests? Just tell me

aivarannamaa, 2019-07-22:
Hmm, this is not the first time when I hear that running from Terminal "cured" Thonny.
I still don't know what to suspect, though...

juanfal, 2019-07-22:
My gosh!  I think Apple it's getting paranoid with security and only Terminal has been able to write(read) necessary for the more controlled Finder open.
Anyway, Aivar, infinite thanks for the best app I've ever seen for teaching how to program. Any further contact with my University, don't hesitate to send me a private message

hroarbye, 2019-10-14:
Upgraded to catalina today (Mac OS 10.15), and getting the same error. Using terminal to launch Thonny works, but no luck getting it working by launching from finder. No luck getting it to work after launching from terminal like juanfal says he did with the beta.

Using Thonny 3.2.1.

aivarannamaa, 2019-10-14:
@hroarbye, can you please check whether the notarized variant works for you - https://github.com/thonny/thonny/releases/download/v3.2.1/thonny-3.2.1.notarized.dmg

hroarbye, 2019-10-14:
Got the same error, but with this version, I works properly AFTER launching with terminal first :) That is, I can doubleclick it to launch after launching once in terminal.

aivarannamaa, 2019-10-14:
@hroarbye, thanks! Can you please do another experiment? 

I'd like to know whether it has problems creating the configuration directory. Please close Thonny, and do `mv ~/Library/Thonny ~/Library/Thonny_temp`. After this try running Thonny again from the icon. It should recreate config directory with default settings and then launch Thonny GUI.

hroarbye, 2019-10-14:
Tried it now, and it worked fine. No errors after launching it fresh.

aivarannamaa, 2019-10-14:
I upgraded my Mac to Catalina and unfortunately I can't reproduce this problem. I didn't experience it with another Mac using Catalina beta either. Maybe it's because of different software configuration.

@hroarbye and @juanfal, did you have official Python 3 distribution (from python.org) installed when you experienced this problem? If yes, then which version?

hroarbye, 2019-10-15:
I did install the official one at one time. Maybe in the middle of august, but then removed it again. I might not have removed all system files, but I do not remember. I used the current version at the time, but do not remember the version number. 3.7.something I guess.

It might be a problem caused by my computer haging during setup of catalina, and having to force-restart the computer to get in. I know other people who have gotten weird errors after upgrading that I cannot reproduce, like the mail app telling them the home folder is full, when they have 300gb avalible and so on. Maybe its a system bug and not a thonny bug. Who knows.

juanfal, 2019-10-15:
Having the same problems (last night 'upgrade' to Catalina 10.15.1 beta (19B68f)). I've been using Thonny with Catalina betas since that last problem until last night 'update'

Terminal can run it, but back to Finder, double click ends with 

    shell-init: error retrieving current directory: getcwd: cannot access parent directories: Operation not permitted

    Backend terminated or disconnected. Use 'Stop/Restart' to restart ...

Apart from the in-the-box pythons, I have brew instalation of 3:

    $ where python3
    lrwxr-xr-x  1 root  wheel  69 22 may  2018 /usr/local/bin/python3 -> ../../../Library/Frameworks/Python.framework/Versions/3.6/bin/python3
    -rwxr-xr-x  1 root  wheel  31456 10 oct 13:39 /usr/bin/python3

juanfal, 2019-10-15:
Notarized versions doesn't fix this for me

rahulswaminathan, 2019-10-15:
> @hroarbye, thanks! Can you please do another experiment?
>
> I'd like to know whether it has problems creating the configuration directory. Please close Thonny, and do `mv ~/Library/Thonny [...]

Hey,

I had the same issue after upgrading to Catalina. But the above quoted experiment fixed the issue. Had to do this even with the notarized image.

Regards

juanfal, 2019-10-15:
For me Terminal execution does it, but double clicking either on the 3.2.1 Thonny icon or on the Notarized Aivaar sent, don't. 

I've had added

    alias thonny='/Applications/Thonny.app/Contents/MacOS/thonny'

to my ~/.bash_profile in the meanwhile

aivarannamaa, 2019-10-15:
> getcwd: cannot access parent directories: Operation not permitted

This is interesting. I'll see whether I need to request more entitlements.

Found a similar issue: https://tex.stackexchange.com/questions/511617/error-with-pdftex-after-updating-to-mac-os-catalina

aivarannamaa, 2019-10-15:
@juanfal, did you try moving or deleting `~/Library/Thonny` ?

tyresedacoldest, 2019-10-15:
@aivarannamaa 

Hi,

I am having the same issue here. I updated my Mac to Catilina 10.15, with Thonny 3.2.1 and when I launch the application it says "Backend terminated or disconnected. Use 'Stop/Restart' to restart ..." in the shell and when I click the stop/restart button or close Thonny, nothing happens. I tried redownloading Thonny entirely, but the issue persisted. 

Any help on this would be greatly appreciated. 
Thank you.

aivarannamaa, 2019-10-16:
I tried another Mac (early 2015 MacBook Air) and still could not reproduce this issue. I installed non-notarized Thonny 3.2.1 to a fresh Mojave, checked that it worked. Then I upgraded to Catalina and could still run Thonny without problems. This time I did not have any 3rd party apps installed beside Thonny.

Then I replaced non-notarized Thonny with notarized variant and could still use it without problems.

Any ideas, what should I change in order to be able to reproduce this problem?

tyresedacoldest, 2019-10-16:
I deleted the data folder ‘Thonny’ that Thonny downloads in the background and reinstalled the application again and it worked.
It seems that the update caused the backend filed to crash, it was downloading thousands of error messages.
I’m not sure if other people are having this issue, but it seems that clearing out everything and reinstalling is the best way to resolve it.

Thanks for taking the Tim eco troubleshoot this.

darkmoondev, 2019-10-16:
> @hroarbye, thanks! Can you please do another experiment?
>
> I'd like to know whether it has problems creating the configuration directory. Please close Thonny, and do `mv ~/Library/Thonny [...]

Was having the same issue on Catalina 10.15 and Thonny 3.2.1, but running this once fixed it for me.

juanfal, 2019-10-16:
Thonny is back.  Don't ask me why/how, I've simply click again on it… perhaps several restarting processes and cache cleaning… I can't imagine the future as it is getting more and more uncontrollable the many layers and layers systems are having
[image]

aivarannamaa, 2019-10-18:
I got my hands on one Catalina where Thonny did not run properly. It was notarized, installed to Desktop and it ran properly only from command line. 

I tried adding "Full disk access" permission to Thonny, but it didn't help. 

After I deleted `~/Library/Thonny`, it didn't show the initial "shell-init" error anymore, but running a script gave a "back-end terminated" error. I then restarted Thonny and tried another script and everything was fine. Finally I ran again the first script and now it ran without problems. After this I didn't get any more errors.

I noticed one difference in the setup compared to my mac -- this one had Desktop under iCloud and I'm not using iCloud. @juanfal, @darkmoondev, @tyresedacoldest, @rahulswaminathan, @hroarbye, where is Thonny located in your computer, under Applications or Desktop? Is connected to iCloud?

rahulswaminathan, 2019-10-18:
Hey @aivarannamaa ,
     On my machine, Thonny is under Applications and not connected to iCloud. I again had the same Backed terminated issue today. Removing the Library/Thonny folder fixed it. One thing I noticed was that the permissions for that Thonny folder are only for my username and no group and other permissions. Could it be a permission issue for a script which initialises the shell? I could be completely wrong but just thought I'll mention it here.

Regards,
Rahul Swaminathan

juanfal, 2019-10-19:
Yesterday I upgraded to last version of catalina beta: 10.15.1 beta (19B77a).  Thonny 3.2.1 is happily starting without any glitches.

- Juan F.

aivarannamaa, 2019-10-25:
I wonder whether deploying Thonny via a pkg installer would make the resulting app-bundle more Gategeeper-friendly.

juanfal, 2019-10-26:
New beta and Thonny is perfectly healthy
[image]

aivarannamaa, 2019-10-31:
@juanfal, @rahulswaminathan, @darkmoondev, @tyresedacoldest -- do you happen to remember, from which directory did run your last script in Thonny before upgrading to Catalina?

When Thonny closes, it saves the directory of last run into its configuration file. When it opens again, the backend process is started in this directory. For some reason, Catalina forbade `os.getcwd()` in this directory. My theory is that once you got Thonny running (either from Terminal or by deleting configuration file and thus resetting working directory to `~/`) you saved a script in this problematic directory and Thonny gained proper access to it.

Can it be that this problematic directory was Desktop or another special directory?

aivarannamaa, 2019-11-01:
When I'm leaving a file in Documents open in Mojave and reopen Thonny again in Catalina, I get

```
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/workbench.py", line 188, in __init__
    self._editor_notebook.load_startup_files()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/code.py", line 614, in load_startup_files
    self.show_file(filename)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/code.py", line 812, in show_file
    editor = self.get_editor(filename, True)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/code.py", line 886, in get_editor
    return self._open_file(filename)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/code.py", line 875, in _open_file
    editor._load_file(filename)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/code.py", line 159, in _load_file
    self._load_local_file(filename, keep_undo)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/code.py", line 172, in _load_local_file
    with open(filename, "rb") as fp:
PermissionError: [Errno 1] Operation not permitted: '/Users/aivar/Documents/kool/kala.py'
```

juanfal, 2019-11-04:
It seems Catalina asks from every app, even Apple's ones, for permission to read many directories, from Desktop to …  Perhaps, the situation is it gains that permission if you open it 

> @juanfal, @rahulswaminathan, @darkmoondev, @tyresedacoldest -- do you happen to remember, from which directory did run your last script in Thonny before upgrading to Catalina?
>
> When Thonny closes, [...]

I am now trying to find how to get rid of "Backend terminated or disconnected. Use 'Stop/Restart' to restart ..."  error in only one out of many fresh new Macs my new first year students are using.  I don't see the difference in the way they have done the things…

Lost

aivarannamaa, 2019-11-04:
> Perhaps, the situation is it gains that permission if you open it

Exactly

> I am now trying to find how to get rid of "Backend terminated or disconnected. Use 'Stop/Restart' to restart ..." error in only one out of many fresh new Macs my new first year students are using. I don't see the difference in the way they have done the things…

Doesn't opening from or saving to the required directory help? In new version I'm catching this error and proposing this workaround in the shell: https://github.com/thonny/thonny/releases/tag/v3.2.3

schellenberg, 2019-11-08:
Can confirm same error on 10.15: 

> `Backend terminated or disconnected. Use 'Stop/Restart' to restart ...`

This occurred when opening a version of Thonny that was installed prior to upgrading to Catalina. Deleting the main Thonny.app bundle and reinstalling did not fix the issue. The fix was to delete BOTH the `~/Library/Thonny` folder, as well as the Thonny.app bundle in Applications, then reinstall completely. Now everything works fine.

juanfal, 2019-11-08:
For the case of my student, stubbornly terminating the backend, it very quickly and happily fixed it to simply take out the ~/Library/Thonny folder while not running

juanfal, 2019-11-08:
Funny, after using it without problems, it can be has happened after I closed every tab on the editor, and after restarting, it is starting to play up:

[backend.log](https://github.com/thonny/thonny/files/3824646/backend.log)

I've patched line 837 of `backend.py` to:

    def send_message(self, msg):
        try:
            if "cwd" not in msg:
                msg["cwd"] = os.getcwd()
        except:
            msg["cwd"] = "/Desktop"

and it seems to work…

juanfal, 2019-11-08:
Now, if anything fails with os.getcwf() it is safely caught:

[image]

and working…  and later, running without problems

[image]

juanfal, 2019-11-12:
It seems that the problem appears with some previous paths saved in `~/Library/Thonny/configuration.ini`, in the var:

    [CustomInterpreter]
    path = …

if you simply clear that variable, deleting it, and rerun Thonny, everything seems to go fine.  I found there weird paths, I didn't know before

hroarbye, 2019-12-02:
I got the error again, running OS 10.15.1.

I have not installed or uninstall anything since it worked fine, and suddenly the error reappeared. Tried clearing with mv ~/Library/Thonny ~/Library/Thonny_temp, and it fixed it again. But thought it would be interresting to know it reappeared without installing a new OS update or anything.

aivarannamaa, 2019-12-02:
@hroarbye, did you start using new (special) directory for your scripts? Documents?  Desktop?

hroarbye, 2019-12-02:
> @hroarbye, did you start using new (special) directory for your scripts? Documents? Desktop?

No. It has just been a few days since I last used it, and when I launced it, I got the error at once. I think maybe my computer had its power cut in the meantime, but nothing out of the ordinary except that. Maybe the restart after losing power did something, but I can't say for sure.

juanfal, 2019-12-26:
V 3.2.4 seems to go fine!
[image]

juanfal, 2019-12-26:
V 3.2.5 Ok!
[image]

allengwinn, 2020-02-05:
I'm having the same issue with 10.15.3. I delete ~/Library/Thonny and delete the application itself. Reinstall it and it works correctly. Reboot the computer, it still works. Come back the next day and the error is back. If I quit and delete ~/Library/Thonny, it works again for another day. Version 3.2.6

aivarannamaa, 2020-02-05:
Please don't give up yet -- a possible solution is coming in couple of days.

I decided to switch from dmg to pkg (installer) and by following the example of Python's official distribution and IDLE, I hope to avoid the problem with permissions in Catalina.

allengwinn, 2020-02-05:
It is possible that the problem could be related to deleting a .py script but not closing it in Thonny. I will try to test it today if I get a chance. A cursory look at the logs showed a getcwd error. I will grant you that I did a whole 45 seconds of poking around before I got distracted.

aivarannamaa, 2020-02-05:
Here comes the installer: https://github.com/thonny/thonny/releases/download/v3.2.6/thonny-3.2.6.pkg (#79)

I have tested it only with Catalina so far. 

Please let me know if this works for you!

juanfal, 2020-02-06:
The download link is broken (404)

aivarannamaa, 2020-02-06:
> The download link is broken (404)

I removed it temporarily because it was broken.

The fixed installer is now available at the same URL: https://github.com/thonny/thonny/releases/download/v3.2.6/thonny-3.2.6.pkg

kcarnold, 2020-02-11:
The `pkg` works for me on macOS Mojave (10.14.6). So no regression for at least this older version of macOS.

nehirkosel, 2020-04-02:
So I have catalina 10.15.3 on my macbook pro and thonny does not seem to be working and it is giving this message no matter what I do  "Backend terminated or disconnected. Use 'Stop/Restart' to restart."   What should I do?

nehirkosel, 2020-04-02:
Ok, I finally solved it and I am posting this so maybe it will help someone. Apparently mine thonny was not up to date, I had to download the new version which was 3.2.7 and that seemed to solve the problem.

aivarannamaa, 2020-12-12:
Haven't heard about this problem for a while. Assuming it is fixed

closed 2020-12-12 by aivarannamaa

juanfal, 2020-12-12:
Thonny on 

	11.1 (20C69) (and previous beta versions)
	MacBook Air (M1, 2020)

like a charm

- Juan Falgueras

aivarannamaa, 2020-12-13:
> Thonny on 11.1 (20C69) (and previous beta versions) MacBook Air (M1, 2020) like a charm.

That's good to know. Thank you!