thonny/thonny #815: Check whether "open system shell" highlights correct pip and python

aivarannamaa, 2019-07-18:


juanfal, 2019-07-18:
It seems they are correctly spotted:

    ********************************************************************************
    Some Python commands in the PATH of this session:
     - python    -> /System/Library/Frameworks/Python.framework/Versions/2.7/bin/python2.7
     - python3   -> /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/bin/python3.7
     - python3.7 == /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/bin/python3.7
     - pip3      -> /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/bin/pip3.7
     - pip3.7    == /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/bin/pip3.7

    ********************************************************************************

juanfal, 2019-07-18:
Ive tried to use other pythons in my system:

    /usr/local/bin/
    /usr/bin/

No luck

aivarannamaa, 2019-07-18:
> Ive tried to use other pythons in my system:
> ...
> No luck

Do you mean same problem as in #813 ?

aivarannamaa, 2019-07-23:
Reminder for myself: test also with virtual env

rpi4iot, 2019-12-25:
in  windows 10 x64   10.0.18363  (windows 10 1909 18363.535)
thonny-3.2.4-windows-portable.zip
**"open system shell" don't display pip and python at all.**
**before a python file opened.** 

**target = shutil.which(cmd)** 
shutil.which  behave differently when a python file open or not.

```
********************************************************************************
Some Python commands in the PATH of this session:
Traceback (most recent call last):
  File "C:\bin\plang\thonny\lib\site-packages\thonny\plugins\system_shell\explain_environment.py", line 118, in <module>
    list_commands("python", [sys_real], highlight_dirs)
  File "C:\bin\plang\thonny\lib\site-packages\thonny\plugins\system_shell\explain_environment.py", line 53, in list_commands
    target = normpath_with_actual_case(target)
  File "C:\bin\plang\thonny\lib\site-packages\thonny\plugins\system_shell\explain_environment.py", line 75, in normpath_with_actual_case
    assert os.path.isabs(name)
AssertionError

C:\bin\plang\thonny>

###################################################
def list_commands(prefix, highlighted_reals, highlighted_dirs):
    for suffix in ["", "3", "3.5", "3.6", "3.7", "3.8"]:
        cmd = prefix + suffix
        print('DBG: ', cmd)
        target = shutil.which(cmd)
        print('DBG: ', target)
###################################################		
		
********************************************************************************
Some Python commands in the PATH of this session:
DBG:  python
DBG:  .\python.EXE
Traceback (most recent call last):
  File "C:\bin\plang\thonny\lib\site-packages\thonny\plugins\system_shell\explain_environment.py", line 120, in <module>
    list_commands("python", [sys_real], highlight_dirs)
  File "C:\bin\plang\thonny\lib\site-packages\thonny\plugins\system_shell\explain_environment.py", line 55, in list_commands
    target = normpath_with_actual_case(target)
  File "C:\bin\plang\thonny\lib\site-packages\thonny\plugins\system_shell\explain_environment.py", line 77, in normpath_with_actual_case
    assert os.path.isabs(name)
AssertionError

C:\bin\plang\thonny>
```

=======================
afte a phython file opened.
it works
```
********************************************************************************
Some Python commands in the PATH of this session:
 - python    == C:\bin\plang\thonny\python.exe
 - pip       == C:\bin\plang\thonny\Scripts\pip.bat
 - pip3      == C:\bin\plang\thonny\Scripts\pip3.bat
 - pip3.7    == C:\bin\plang\thonny\Scripts\pip3.7.bat

********************************************************************************

C:\bin\plang\thonny\prjs\prj0>
```

aivarannamaa, 2019-12-25:
Thank you for the report and debugging!

It is surprising that `shutil.which` can return relative path, but it  looks like this is the case when the target is in current directory. I'll fix this assumption. 

Besides this, I'll change the way how current directory for the terminal will be selected. At the moment it uses dir of active file, but Thonny actually also maintains clear notion of current directory. This directory can be changed either by Files view or by running a script when auto-cd is enabled (Tools => Options => Run & Debug).

closed 2019-12-25 by aivarannamaa