thonny/thonny #828: Raspberry Pi 4 thonny will run under raspbian's normal desktop but not under mate

MindMapMaker, 2019-07-30:
Raspberry Pi 4: thonny will run under raspbian's normal desktop but not under mate desktop. 
I need mate desktop because I need a dark theme for my eyes.
 
os: buster installed from noobs on July 28, 2019
Thonny 3.1.0 as installed as part of the "complete system" install

On launch Thonny reports the following then exits:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/__init__.py", line 65, in launch
    bench = workbench.Workbench(server_socket)
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 145, in __init__
    self._load_plugins()
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 340, in _load_plugins
    self._load_plugins_from_path(thonnycontrib.__path__, "thonnycontrib.")
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 360, in _load_plugins_from_path
    getattr(m, load_function_name)()
  File "/usr/lib/python3/dist-packages/thonnycontrib/pi/__init__.py", line 248, in load_plugin
    update_fonts()
  File "/usr/lib/python3/dist-packages/thonnycontrib/pi/__init__.py", line 227, in update_fonts
    fontname = re.search(r"=([^0-9]*) ([0-9]*)", line, re.M).group(1)  # @UndefinedVariable
AttributeError: 'NoneType' object has no attribute 'group'

aivarannamaa, 2019-07-30:
Thank you for the report!

I believe this will be fixed in next version (3.2). I'll check later.

In the meanwhile the workaround is to uninstall `thonny-pi` package (you would lose RPi theme and some minor settings will change).

aivarannamaa, 2019-07-30:
Can you please send me the names and contents of your desktop config files that Thonny inspects? Following commands will print the required file paths:

```
echo "~/.config/lxsession/${DESKTOP_SESSION}/desktop.conf"
echo "/etc/xdg/lxsession/${DESKTOP_SESSION}/desktop.conf"
```

closed 2021-02-27 by aivarannamaa