thonny/thonny #829: Basic Function in Python

saidor54, 2019-07-30:
I created consecutively functions that return the values by putting them in the title form. However, after running the code, there occurs some extra string which is "none". I inserted the code below. How do I fix this? Thank you.

def make_title():
    x = input("What is the title?: ")
    print(title_it(x))
    
>>> make_title()
What is the title?: said
Said
None
>>> def make_title():
    tit = input("What is the title?: ")
    print(title_it(tit))
    
>>> make_title()
What is the title?: said
Said
None
>>> def title_it_rtn():
    #rtn = make_title()
    return make_title()

>>> title_it_rtn()
What is the title?: merhaba canım
Merhaba Canım
None
>>>

saidor54, 2019-07-30:
[image]

aivarannamaa, 2019-07-30:
You should ask this in some Python beginner forums. It might be even good question for Stackoverflow.

This issue tracker is only meant for problems related to Thonny IDE.

closed 2019-07-30 by aivarannamaa