thonny/thonny #833: Saving is impossible

Jpn287, 2019-08-01:
Hello.
When I try to save my python file this pop out :
[image]

> _File not found. Verify it's name and try again_

But I'm only trying to save as, not to find a file...
Can you help ? ^^

aivarannamaa, 2019-08-01:
Hi!

That's strange. Do you have any special characters in your user name? Can you save outside of your home directory? Can you save directly into your home directory (ie. the parent directory of Documents)?

Jpn287, 2019-08-04:
No same username than on Git. I tried to save on desktop and I will try in the parent directory. I will come back on this after that.

Jpn287, 2019-08-14:
Hi, @aivarannamaa.
Yes **saving into the parent folder solved the problem**. I don't understand what caused this dysfunction.
Cordially.

aivarannamaa, 2019-09-01:
@Jpn287, can you please post the output of following program:

```
import os.path

home = os.path.expanduser("~")
print("Home:", home)
print("Documents exists:", os.path.exists(os.path.join(home, "Documents")))
print("Desktop exists:", os.path.exists(os.path.join(home, "Desktop")))

```

Jpn287, 2019-09-02:
### Here it is :
```Traceback (most recent call last):
  File "C:\Users\japan\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\japan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\code.py", line 601, in _cmd_save_file
    self.get_current_editor().save_file()
  File "C:\Users\japan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\code.py", line 240, in save_file
    f.flush()
OSError: [Errno 9] Bad file descriptor
```

Thanks for your hard work

Jpn287, 2019-09-02:
When launching into Thonny : 
```
Home: C:\Users\japan
Documents exists: True
Desktop exists: True
```
And when trying to save : 

``` 
Traceback (most recent call last):
  File "C:\Users\japan\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\japan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 819, in dispatch_from_menu
    dispatch(None)
  File "C:\Users\japan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 777, in dispatch
    handler()
  File "C:\Users\japan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\code.py", line 613, in _cmd_save_file_as
    self.get_current_editor().save_file(ask_filename=True)
  File "C:\Users\japan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\code.py", line 240, in save_file
    f.flush()
OSError: [Errno 9] Bad file descriptor
```
Thanks for your hard work.

aivarannamaa, 2019-09-03:
Do you have a single user PC, or a classroom computer with shared disks or something like this?

Jpn287, 2019-09-03:
Only a single user PC W10.

aivarannamaa, 2019-09-14:
At the moment I don't have any new ideas how to reproduce it.

Jpn287, 2019-09-14:
Si for now I will close the thread because for now it seems to be a unilateral case. Thanks for all if I get any news I will come back.

aivarannamaa, 2019-09-15:
It's OK if we leave it open. I'll just move it to next milestone when required.

aivarannamaa, 2019-10-03:
Can you write to your Documents using following Python script in Thonny (can be pasted to Thonny's shell, if you can't save it to a script):

```
f = open(r"C:\Users\japan\Documents\test.txt", "w")
f.write("testing")
f.close()
```

?

aivarannamaa, 2020-03-29:
@Jpn287 , did you find a solution?

Jpn287, 2020-03-29:
> @Jpn287 , did you find a solution?

No but I thought that because I’m not using Thonny I should close the issue. Is it okay ?

aivarannamaa, 2020-03-29:
I'll leave it open so I can find it more easily when same problem pops up again.