thonny/thonny #834: Fails to start with Tk older than 8.6

kapsh, 2019-08-03:
Thonny fails to start when Tcl/Tk version is below 8.6 with error: 
```
Traceback (most recent call last):
  File "/home/kapsh/.local/pipx/venvs/thonny/lib/python3.7/site-packages/thonny/__init__.py", line 65, in launch
    bench = workbench.Workbench(server_socket)
  File "/home/kapsh/.local/pipx/venvs/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 185, in __init__
    self._publish_commands()
  File "/home/kapsh/.local/pipx/venvs/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 751, in _publish_commands
    self._publish_command(**cmd)
  File "/home/kapsh/.local/pipx/venvs/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 822, in _publish_command
    _image = self.get_image(image)  # type: Optional[tk.PhotoImage]
  File "/home/kapsh/.local/pipx/venvs/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 1369, in get_image
    img = tk.PhotoImage(tk_name, file=filename)
  File "/usr/lib/python3.7/tkinter/__init__.py", line 3545, in __init__
    Image.__init__(self, 'photo', name, cnf, master, **kw)
  File "/usr/lib/python3.7/tkinter/__init__.py", line 3501, in __init__
    self.tk.call(('image', 'create', imgtype, name,) + options)
_tkinter.TclError: couldn't recognize data in image file "/home/kapsh/.local/pipx/venvs/thonny/lib/python3.7/site-packages/thonny/res/run-current-script.png"
```

PNG format is supported since Tk 8.6 https://docs.python.org/3/library/tkinter.html?#images so it would be friendlier to check `tkinter.TkVersion` value before trying to show GUI and shutdown gracefully saying to update dependencies.

aivarannamaa, 2019-08-03:
Thank you for the report!

I'll add this check and notification.

aivarannamaa, 2020-12-12:
Closing, because I assume Tk 8.5 is rare enough by now.

closed 2020-12-12 by aivarannamaa