thonny/thonny #835: How to build executable installer?

Xbeninni, 2019-08-05:
I am sorry to bother you guys. I know this is not actually an issue but questions from Python beginner about Thonny. Any help will be much appreciated! My questions are, 

**1) How Thonny build executable installer?** 
I know Pyinstaller is an option to build executable, but the way Thonny use seems more professional.

**2) Is tkinter the only UI library that Thonny use?**

Thank you!

aivarannamaa, 2019-08-09:
Hi!

1) Thonny is not anyway related to PyInstaller or any of the alternatives. It does not have any special support for building executables.
2) Thonny is built with Tkinter, but you can use any GUI framework in your programs (if you install required packages).

Xbeninni, 2019-08-09:
@aivarannamaa  thank you for your answer, I appreciate it! I am sorry if any misunderstanding, but I am not seeking support for building executable. I just wondering technology does Thonny use to build executable

aivarannamaa, 2019-08-09:
Thonny uses official Python executable + stdlib + 3rd party libraries.

In Windows it has a custom wrapper exe for invoking `thonny` module (https://github.com/thonny/thonny/tree/master/packaging/windows/ThonnyRunner37). All is tied together with Inno Setup (https://github.com/thonny/thonny/blob/master/packaging/windows/inno_setup.iss)

aivarannamaa, 2019-08-09:
In Windows I'm installing official Python installation, copy it to `C:\Pythonny37` and run this: https://github.com/thonny/thonny/blob/master/packaging/windows/create_installer.bat

Xbeninni, 2019-08-10:
Thank you @aivarannamaa , it was clear and helpful!

closed 2019-08-10 by Xbeninni