thonny/thonny #836: Add option to automatically soft-reboot micropython after uploading scripts

Ivoz, 2019-08-05:
What it says on the title.

At least for my esp32, it requires a soft-reboot for the newly reloaded code to automatically run. I imagine though there may be cases where you simply want the new file uploaded but no restart, so a configuration option would be nice.

aivarannamaa, 2019-08-05:
I assume you want it for quick testing. Isn't "Run current script" suitable for this?

Ivoz, 2019-08-05:
My code is not entirely in one file, so not all files are "scripts" per say

aivarannamaa, 2019-08-05:
I'll propose a more general solution -- Replace "Run current script" with simply "Run" and add an easy option to specify the main script to be run (eg. as combobox on the toolbar). Default value would be `<current script>`.

BTW. Have you checked out 3.2 beta? It has many changes in MicroPython support.

Ivoz, 2019-08-05:
ATM I was re-uploading my module file, then rebooting (button or with ctl-D) my esp32 so it ran `main.py` again which would cleanly reimport the new modules with the changes.

If running `main.py` again would also result in a clean re-import of the new code then that would be fine.

I had switched for a little to a new Fedora install, and was running their repo's default 3.1.1 which unfortunately has a stalling / completely locks up on quit issue ATM :/

aivarannamaa, 2020-02-14:
Will be dealt with under #1087

closed 2020-02-14 by aivarannamaa