thonny/thonny #840: Files fail to open or save (Mac OS X 10.15)

troywein, 2019-08-08:
Hello I am encountering a weird issue where I am unable to load or save python scripts.  I recently upgraded to Mac OS X 10.15.  I have tried reinstalling and also have tried running the application from different install locations.  Thanks Troy

aivarannamaa, 2019-08-08:
Hi!
Do you get some error messages? How does the problem manifest?

troywein, 2019-08-08:
Upon double clicking any file.py it doesn’t open in the editor.  On save after clicking save the dialog closes and the open file remains as <untitled> *.  I’ve attached a screenshot.  Thanks Troy

[cid:3694E61E-156D-499C-BA07-6403284EE72F]

aivarannamaa, 2019-08-09:
> Upon double clicking any file.py it doesn’t open in the editor

Do you mean double clicking in Finder or in Thonny's Files view?

Can you try running Thonny from Terminal? It may give some clue. Logs may be useful as well. The instructions are here: https://github.com/thonny/thonny/wiki/Troubleshooting#open-in-terminal-macos

troywein, 2019-08-09:
This issue only happens on my machine running the latest beta of Mac OS X.  I was able to test on an older version of Mac OS and the issue isn’t there.  Do you have access to a machine with the latest beta?   I can check the log files and send them to you if you are interested?   We could also wait and see if the issue persists in the next release.

You are correct a double clicked file will not open.

Thx Troy

aivarannamaa, 2019-08-12:
> I can check the log files and send them to you if you are interested?

Yes, this could be helpful!

> Do you have access to a machine with the latest beta?

My Mac isn't compatible, but I'll try to install it in a VM.

troywein, 2019-08-12:
Aivar,

I’ve attached a screenshot of the terminal window output when running Thonny from Terminal.  I first attempted to double click a dot.py file and then attempted to save.  The logs are also attached.  Thx Troy
[cid:0869ea4c-bc31-4041-ae7f-9b93d3bc3f4d@namprd17.prod.outlook.com]

aivarannamaa, 2019-08-13:
Unfortunately I didn't receive the attachments. It seems to be usual problem of GitHub that the attachments sent by e-mail get lost. Please open this issue in the browser and upload the attachments via GitHub interface!

troywein, 2019-08-14:
Here are the files.
[image]
[Thonny.zip](https://github.com/thonny/thonny/files/3502052/Thonny.zip)

aivarannamaa, 2019-08-14:
Do you happen to have separate Python + IDLE installed?

If yes, then please check whether following program works for you:

```
import tkinter as tk
from tkinter import filedialog

def doit(event):
    print(filedialog.askopenfilename(master=root))

root = tk.Tk()
root.bind("<1>", doit, True)

root.mainloop()
```

When you run it then you should see an empty window.  If you click on it, you should see File Open dialog. Please test whether deleting `master=root` from the code changes anything.

aivarannamaa, 2019-08-14:
Can you open other Thonny's dialogs (eg. Tools => Options or Help => About) ?

troywein, 2019-08-14:
I can open the dialogs.

troywein, 2019-08-14:
> Do you happen to have separate Python + IDLE installed?
>
> If yes, then please check whether following program works for you:
>
> ```
> import tkinter as tk
> from tkinter import filedialog
>
> def [...]

troywein, 2019-08-14:
I do have a separate Python and Idle installed.  I get the following error message when running the program you have above:

[image]

aivarannamaa, 2019-08-14:
Most likely `python` command invokes Python 2 bundled with the OS. Can you try same program with `python3`?

troywein, 2019-08-14:
Works fine in python3 with and without master=root

closed 2019-08-28 by aivarannamaa

aivarannamaa, 2019-08-29:
Problem was created in 4db8c4846793b66eaa473e9f4bdf6f9b180ab165

commit 27692a75 (2019-08-28): FIXED #840: Problems with file dialogs in macOS 10.15