thonny/thonny #847: http://localhost:7777/ipython_call rejected conection

skoll43, 2019-08-11:
im extra virgin in programming haha and using on thonny, the thing is, the program open a browser windows and then, dont show how is supposed to, i even make:
0. my private ip dmz, turn down smartscreen (windows)
1 .antivirus defender on all his options
2. make exclusions for firewall, deactivated the firewall, changed the port to various numbers
3. open manually the ports on router, i mean, how unsecure my notebook is supposed to be?! 
   hahaha 

> [image]

packages in thonny: 
0. [image]
1. 
[image]

the bottom of this, is that im trying to understand this code:
```
def tri_recursion(k):
  if(k>0):
    result = k+tri_recursion(k-2)
    print(result)
  else:
    result = 0
  return result

print("\n\nRecursion Example Results")
tri_recursion(6)
```

any help will be greatly appreciated, 
opened first on https://github.com/alexmojaki/birdseye/issues/77
on normal installation of python (without thonny) works normal

aivarannamaa, 2019-08-13:
I could reproduce this problem when running from installed Thonny, but everything worked fine when run in my development envrionment. It seems that one of the transient dependencies of Birdseye (click) can fail when it is run without console:

```
Traceback (most recent call last):
  File "C:\Users\Aivar\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\birdseye_frontend.py", line 23, in start_server
    port=get_workbench().get_option("run.birdseye_port"), debug=False, use_reloader=False
  File "C:\Users\Aivar\AppData\Local\Programs\Thonny\lib\site-packages\flask\app.py", line 985, in run
    cli.show_server_banner(self.env, self.debug, self.name, False)
  File "C:\Users\Aivar\AppData\Local\Programs\Thonny\lib\site-packages\flask\cli.py", line 670, in show_server_banner
    click.echo(message)
  File "C:\Users\Aivar\AppData\Local\Programs\Thonny\lib\site-packages\click\utils.py", line 260, in echo
    file.write(message)
AttributeError: 'NoneType' object has no attribute 'write'
```

I was able to work around this problem in Thonny's back-end, but for some reason the same trick didn't work for the front-end (which launches Birdseye server).

@alexmojaki , you are probably more knowledgeable with Flask and its ecosystem -- is it possible to force `click` to non-tty mode?

alexmojaki, 2019-08-13:
Sorry, I've never touched click before. So is the problem that it's trying to write to stdout/stderr? Why is it None?

aivarannamaa, 2019-08-13:
I think it gets confused with Thonny's approach of allocating console. I created an issue: https://github.com/pallets/click/issues/1367

aivarannamaa, 2020-04-12:
Fixed under #1141

closed 2020-04-12 by aivarannamaa