thonny/thonny #852: Launch problem in Ubuntu

akaioda, 2019-08-13:
HI,
I've tryed to install thonny on an ubuntu 18.04 desktop.

First with the ubuntu app installer, then with the bash <(curl -s https://thonny.org/installer-for-linux) command.

In both case when i click on the thonny icon in the app laucher the application doesn't show on, until i depress ctrl or alt key for exemple or evently if I close an other windows on the desktop.

Any idea to fix this ? because thonny looks great en i've already enjoy it under windows 10.

I'm a newby to ubuntu an i've no direction to troobleshoot this issue.
Of course i'll will also post this question in Ubuntu help forum.

Best regards

akaioda, 2019-08-13:
Hello,

Thank you to take in account !

At this time i've install thonny using the bash command.

So when i use ~/apps/thonny/bin/thonny, nothing appears ... the same as by using the launcher !
Then as said, by depressed the ctrl key the app show on.
After closing no error appears in the terminal.

Anyway, i confirm that after launching with the tricky way, thonny works as expected ! 

Just that it will be more elegant and classical if it works by start by a single click :-)

regards

aivarannamaa, 2019-08-13:
That's strange. I can't reproduce this on my Ubuntu 18.04.

Have you installed some kind of desktop extensions or something like that?

akaioda, 2019-08-13:
Argh ... Thanks again ! !

No, if i understood you, nothing else than other applications !

using this cookbook : 
https://askubuntu.com/questions/1133782/command-to-list-installed-and-enabled-gnome-extensions

may confirm my talk !? :

1- /home/[myusername]/.local/share/gnome-shell/extensions/ 
**does not exist**
2- gsettings get org.gnome.shell enabled-extensions
**return : as []
I think it means nothing !?**
3- ls /usr/share/gnome-shell/extensions/
**return : ubuntu-appindicators@ubuntu.com  ubuntu-dock@ubuntu.com
wich i think to be launcher side bar and top bar indicator, so buitin and  mandatory !?**

I've install log viewer application but see nothing special related to thonny excepted :
 
11:50:49 thonny: Delegated to an existing Thonny instance. Exiting now.
**I thing this kind of line, is due to single instance control of thonny you probably apply**
11:04:39 dbus-daemon: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.130' (uid=1000 pid=5248 comm="zenity --name=Thonny --class=Thonny --file-selecti" label="unconfined")
11:04:00 thonny: Delegated to an existing Thonny instance. Exiting now.

So i don't see the light :-\

Regards

aivarannamaa, 2019-10-11:
Can you test whether version 3.2.1 works for you?

closed 2021-01-19 by aivarannamaa