thonny/thonny #854: Unable to use and information leak on multi-user systems

Natureshadow, 2019-08-14:
We tried using Thonny in an educational environment with a remote desktop server used by multiple users in parallel, to find out that this is jut not possible currently.

When starting Thonny, it will find the open TCP socket of the instance run by another user, and connect to it.

This causes THonny to not start for more than one user, and also leaks any action triggered on the command-line to the instance of the other user.

Please use UNIX domain sockets in a private directory of the user running thonny instead, and only fall back to TCP on WIndows (if you really need to support this platform).

aivarannamaa, 2019-08-14:
> Please use UNIX domain sockets in a private directory of the user running thonny

Thanks for the tip! I'll will look into this.

As a workaround I suggest to turn off single-instance feature. In 3.2 it's "Tools => Options => General => Allow only single Thonny instance"

> only fall back to TCP on WIndows (if you really need to support this platform).

Yes, I really need to support this platform -- about 80% of Thonny's non-Raspberry users are on Windows :)

Natureshadow, 2019-08-14:
>As a workaround I suggest to turn off single-instance feature. In 3.2
>it's "Tools => Options => General => Allow only single Thonny instance"

Not so easy without starting Thonny first ;).

Why is this "feature" there, by the way? I can see no benefit in it.

>> only fall back to TCP on WIndows (if you really need to support this
>platform).
>
>Yes, I really need to support this platform -- about 80% of Thonny's
>non-Raspberry users are on Windows :)

So then maybe just turn off single-instance there (or everywhere, which would be my favourite solution ;)).

Ivoz, 2019-08-15:
> Why is this "feature" there, by the way? I can see no benefit in it.

If you open up multiple code files from a file explorer but don't want multiple appplication windows resulting, only new tabs in a single application?

Natureshadow, 2019-08-15:
Hmm, okay. But don't share a socket across multiple users for that!

By the way, it seems that current Windows 10 indeeed does support AF_UNIX. So it seems the only hard part is finding a user-specific directory to put the socket in. I suggest the following:

On UNIX:

1. If `$XDG_RUNTIME_DIR` is set, use `$XDG_RUNTIME_DIR/thonny/ipc.sock`
2. If `$TMPDIR` is set, use `$TMPDIR/thonny-$USER/ipc.sock`
3. Else, use `/tmp/thonny-$USER/ipc.sock`.

Be sure to create the sub-directory with mode 0700.

On WIndows:

1. Use `%appdata%\thonny\ipc.sock`.

aivarannamaa, 2019-08-24:
As AF_UNIX is not supported by Python on Windows (yet), I created another solution. Instead of using fixed port number, first Thonny instance run by a user picks a random dynamic port and publishes it under a lock file inside Thonny user directory. Next instances pick up the port number and use it for forwarding their requests.

TODO: I still need to test the solution in Linux and Mac and when single instance mode is turned off.

Natureshadow, 2019-08-24:
> As AF_UNIX is not supported by Python on Windows (yet), I created
> another solution. Instead of using fixed port number, first Thonny
> instance run by a user picks a random dynamic port and publishes [...]

This does not solve the problem - you'd still have to add proper authentication, at the very least.

You can use this workaround on Windows, but please stop crippling Thonny on proper OSs.

aivarannamaa, 2019-08-25:
> This does not solve the problem - you'd still have to add proper authentication, at the very least.

IMO it does solve the core of the problem. I suppose you meant this is not a perfect solution for this problem.

It can be easily secured by setting suitable permissions for the lock file and making server publish a secret that must be passed along with each request.

> [...] please stop crippling Thonny on proper OSs

Please explain, in which way this is crippling Thonny?

Aside: there are about 3x more Thonny users on Windows 7 (which will probably never get AF_UNIX support) than on "proper OSs" (I assume you meant Linuxes).

Natureshadow, 2019-08-25:
>IMO it does solve the core of the problem. I suppose you meant this is
>not a perfect solution for this problem.

It *works around* the core of the problem.

But thinking about it, it introduces a new problem: It breaks using the same home directory on several computers simultaneously, like guest accounts on some educational environments do (yes, this is also not nice...). So you would need to make sure to track multiple entries, distinguished by host,...

>It can be easily secured by setting suitable permissions for the lock
>file and making server publish a secret that must be passed along with
>each request.

That would be one step, yes.

>Please explain, in which way this is crippling Thonny?

You are re-inventing the wheel. And you are not the first to do so. Mozilla, for one, has gone through this, at least the lock file idea, and abandoned it later because it made too many assumptions about where Firefox was run.

As a side note, Thonny is educational software, and thus should, next to being a good tool, set an example for how to do things right. That includes choosing the right technologies instead of re-inventing the wheel.

A ton of people have explored how to do IPC, and agreed on standards, like the availability of XDG_RUNTIME_DIR that specifically solves all of the problems we face here.

Use standards. (They can't be so unnecessary if even Microsoft feels motivated to copy them, can they ;)?)

>Aside: there are about 3x more Thonny users on Windows 7 (which will
>probably never get AF_UNIX support) than on "proper OSs"

Windows 7 is end of life. We really should stop helping everyone keep their outdated, unmaintained systems.

But I see where you come from here - thus: Use the right tool if available, and only fall back to workarounds if they are not available.

> (I assume you meant Linuxes).

Actually mostly anything but Windows NT. Linux has its issues as well. Being unix-like comes closer.

aivarannamaa, 2019-08-26:
> it introduces a new problem: It breaks using the same home directory on several computers simultaneously

Wasn't it the case with previous solution (as in 3.2.0) as well?

Now Thonny instances are communicating in groups tied to the same configuration directory and single instance mode can be turned off more easily. Sharing configuration directories between different actual users will cause problems anyway.

> Use the right tool if available, and only fall back to workarounds if they are not available.

I fail to see any practical problems with current approach. It's the simplest solution I could design for Windows. As it works in Linux/Mac as well, I don't see the need to complicate code by adding alternative implementation for these OSs.

Natureshadow, 2019-08-26:
>> it introduces a new problem: It breaks using the same home directory
>on several computers simultaneously
>
>Wasn't it the case with previous solution (as in 3.2.0) as well?

No. It used port 7777 there, and opened that on the machine if it wasn't open.

With your proposed solution, it would do the following:

1. Start thonny on machine A. It gets a random port.
2. Start thonny on machine B. It finds the lock file, but will fail to connect to the port.

Now it can do one of two things:

3a. Get another random port and start thonny there, writing that port to the lock file. This will break the running thonny on machine A

3b. Not start.

>I fail to see any practical problems with current approach. It's the
>simplest solution I could design for Windows. As it works in Linux/Mac
>as well, I don't see the need to complicate code by adding alternative
>implementation for these OSs.

It works unless you hit some edge case. Oh, by the way, some terminal server solutions don't allow opening random IP server sockets to users at all...

I see no reason for a workaround that will break under some circumstances (and I am in touch with setups that are broken by either of your solutions - this is not some made up theory, but I, for one, *will* *practically* have to ditch thonny at a part of the schools I work with), while there is a real and viable solution.

aivarannamaa, 2019-08-26:
> 3a. Get another random port and start thonny there, writing that port to the lock file. This will break the running thonny on machine A

Server-Thonny doesn't care about the lock file after it has written to it until shutdown, when it tries to remove it (I just added a [fix](https://github.com/thonny/thonny/commit/e16da33b687359b5a92f4aafd7aaf56975ca30d8) for this). I haven't tested it, but I don't see how machine B overwriting the lock file can disturb machine A.

I understand, that more complete and elegant solution exists for Linux and Mac, but I can't use it in Windows and I'm hesitant to complicate the code with two solutions unless it brings noticeable benefits.

As I see it, the only problem with current solution (from the user perspective) is that in setups with shared configuration directories, single instance mode does not work properly (new instances appear when another user overwrites the lock file). IMO this is rather small problem compared to sharing configration with other users.

Natureshadow, 2019-08-26:
> As I see it, the only problem with current solution (from the user
> perspective) is that in setups with shared configuration directories,
> single instance mode does not work properly (new instances [...]

Guest account...

That, and setups where users cannot open IP server sockets.

What excites me is that we both have the same problem in practice, but you only see your side. You need to support strange setups (Windows), and I have to support strange setups (terminal server environments with limitations). We both have no influence on that, but we could easily make thonny work on both.

Natureshadow, 2019-08-26:
We could also use named pipes instead. They work on all platforms, with the same code (except for the path).

aivarannamaa, 2019-08-26:
Are you sure there don't exist strange Linux or Mac setups where AF_UNIX solution doesn't work properly?

aivarannamaa, 2019-08-26:
> We could also use named pipes instead. They work on all platforms, with the same code (except for the path).

I considered it, but it's quite complex in Windows.

Natureshadow, 2019-08-26:
>Are you sure there don't exist strange Linux or Mac setups where
>AF_UNIX solution doesn't work properly?

In theory, yes. But no desktop environment, no ssh, no gpg, no pulseaudio, you name it would work there either, so no, in practice, there is no such setup.

aivarannamaa, 2019-08-26:
I don't know how to determine whether a pipe is operational (ie. server is reading it) or it is abandoned by it's creator (eg. the process has been killed).

aivarannamaa, 2019-08-27:
I added AF_UNIX and XDG_RUNTIME_DIR usage for non-Windows systems.

TODO: Test in macOS.

aivarannamaa, 2019-08-29:
Seems to work in macOS

closed 2019-09-01 by aivarannamaa