thonny/thonny #859: Single instance fails with exception

Natureshadow, 2019-08-15:
On my system, trying to open another file (`setup.py` here) in a running instance from the command-line produces the following exception:

```
Traceback (most recent call last):
  File "/usr/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/usr/lib/python3.7/tkinter/__init__.py", line 749, in callit
    func(*args)
  File "/home/nik/.local/lib/python3.7/site-packages/thonny/workbench.py", line 2044, in _poll_socket_requests
    self.get_editor_notebook().show_file(filename)
  File "/home/nik/.local/lib/python3.7/site-packages/thonny/code.py", line 831, in show_file
    editor = self.get_editor(filename, True)
  File "/home/nik/.local/lib/python3.7/site-packages/thonny/code.py", line 882, in get_editor
    return self._open_file(filename)
  File "/home/nik/.local/lib/python3.7/site-packages/thonny/code.py", line 871, in _open_file
    editor._load_file(filename)
  File "/home/nik/.local/lib/python3.7/site-packages/thonny/code.py", line 162, in _load_file
    self._load_local_file(filename, keep_undo)
  File "/home/nik/.local/lib/python3.7/site-packages/thonny/code.py", line 182, in _load_local_file
    filename = normpath_with_actual_case(filename)
  File "/home/nik/.local/lib/python3.7/site-packages/thonny/common.py", line 219, in normpath_with_actual_case
    assert os.path.isabs(name) or os.path.ismount(name), "Not abs nor mount: " + name
AssertionError: Not abs nor mount: setup.py
```

Even without the changes in #857.

aivarannamaa, 2019-08-17:
Did you invoke something like this: `thonny setup.py`, ie. with relative path?

I guess I need to update the code such that it transforms relative paths to absolute before passing them to existing Thonny instance.

Natureshadow, 2019-08-17:
>Did you invoke something like this: `thonny setup.py`, ie. with
>relative path?

Yes.

commit d8e09eb3 (2019-08-18): FIXED #859: Support relative paths on command line
closed 2019-08-18 by aivarannamaa