thonny/thonny #862: Request for 64-bit Python

jr751, 2019-08-16:
I want to be able to use modules like tensorflow and TFLearn with Thonny, but it doesn't work. I think the reason why is because they need the 64-bit version of Python. But Thonny only comes with the 32-bit version of Python. I would like to request that you modify Thonny so that it comes with 64-bit Python instead. That we can use some of these commonly used modules.

aivarannamaa, 2019-08-18:
I can't just replace 64-bit with 32-bit, because Thonny is also being used on 32-bit computers. I'll consider adding 64-bit variant later.

I recommend you install separate 64-bit Python and let Thonny use it (Tools => Options => Interpreter)

jr751, 2019-11-14:
No, this really doesn't solve the problem either. For example, you can do as you suggested, but then for example, if I want to use Matplotlib I run into problems. You can install Matplotlib and its dependencies, however, when you then try to run it by typing 'import matplotlib.pyplot as plt' you get the error message "OSError: [WinError 193] %1 is not a valid Win32 application". So why isn't this working?

aivarannamaa, 2019-11-14:
I can't reproduce this. 

I used Thonny 3.2.3 Windows bundle (ie. with 32-bit Python 3.7.5). I installed separately 64-bit Python 3.8 from python.org and assigned it as Thonny's back-end (Tools => Options => Interpreter => Alternative Python 3 interpreter). Then I installed matplotlib from Tools => Manage packages (Had to select 3.2.0rc1 from "..." button, because latest stable doesn't provide binaries for Python 3.8).

After this I could run the example code: 

```
import matplotlib.pyplot as plt
plt.plot([1, 2, 3, 4])
plt.ylabel('some numbers')
plt.show()
```

jr751, 2019-11-15:
No, still not working for me and giving me the same error. The only thing I am doing different from you is using Python 3.7.5 (64-bit). I can't use Python 3.8 because it won't let you install the grpcio module, and I need that dependency if I am to user tensorflow. I switched the matplotlib version to 3.2.0rc1. Same error message though. As for its dependencies, I am using numpy 1.17.0, setuptools 41.1.0, cycler 0.10.0, python-dateutil 2.8.1, kiwisolver 1.0.1.

aivarannamaa, 2019-11-16:
> when you then try to run it by typing 'import matplotlib.pyplot as plt' into the python shell you get the error message "OSError: [WinError 193] %1 is not a valid Win32 application"

I would expect this when you somehow get 64-bit matplotlib into 32-bit python. Please check whether you notice something suspicious in the output of following script:

```
import sys
import platform

print("ARCH:", platform.architecture())
print("EXE:", sys.executable)
print("PATH:", sys.path)
```

pssolanki111, 2020-11-21:
@aivarannamaa Hey Aivar, sorry for troubling you again. I was wondering which version (or rather edition) of thonny should I use for 64 bit Windows 10. I just tried installing [THIS](https://github.com/thonny/thonny/releases/download/v3.3.0/thonny-3.3.0.exe) one but it doesn't get installed.

aivarannamaa, 2020-11-21:
@pssolanki111, this should work both in 32 and 64-bit Win 7 and Win 10.

What do you mean by not getting installed? Did you get an error message?

pssolanki111, 2020-11-21:
@aivarannamaa  Yes, I did get an error saying

> This app can't be installed in your computer. Contact the app vendor to get a compatible version.

(It wasn't the exact same message as above but it meant somewhat the same)

aivarannamaa, 2020-11-21:
Do you have "regular" PC, ie. with Intel or AMD processor? Not ARM?
In this case I recommend to download it again -- maybe the download was broken first time.

pssolanki111, 2020-11-22:
@aivarannamaa Yes. It's AMD Ryzen 3. I'll try re-downloading.

pssolanki111, 2020-11-22:
It did work this time, Thanks :)

aivarannamaa, 2021-12-26:
The main Windows installer will come with 64-bit Python 3.10 starting with Thonny 4.0

closed 2021-12-26 by aivarannamaa

pssolanki111, 2021-12-26:
@aivarannamaa great. thonny helped me a lot when i started out and i recommend it to anyone I can :) 

just wanted to say thanks for the wonderful tool.