thonny/thonny #863: how to clear recent file

LGCarmonaS, 2019-08-17:
I have been unable to clear the recent file list, is there a way to do this? Thanks

aivarannamaa, 2019-08-18:
Nope.

Why do you want to clear it?

LGCarmonaS, 2019-08-18:
Thank you for your quick responce. The reson is simple, it has become very long. I only work on  the last few files thus, as it becomes crowded, I wanted to clear and start fresh as I move along my learnibg. Thank you again. Something to consider in the future. Great IDE!!!

LGCarmona

----- Mensaje original -----

Fecha: Sábado, 17 de Agosto de 2019, 22:51
A: thonny/thonny <thonny@noreply.github.com>
CC: LGCarmonaS <luis.carmona@costarricense.cr>, Author <author@noreply.github.com>

  ( ">    LGCarmona
,(> )>
(  ^^    luis.carmona@costarricense.cr
Todos tenemos opiniones y creencias y las expresamos todo el tiempo, pero, ¿serán realmente ciertas o sólo es nuestro ego el que habla? ~ Anónimo

aivarannamaa, 2019-08-24:
I'd rather not clutter the interface with another command. The recent files list starts removing older entries automatically once it sees more than 15 files.

closed 2019-08-24 by aivarannamaa