thonny/thonny #869: Suffestion:render whitespace

Dleta, 2019-08-25:
I think it's useful to see whitespace characters (space, tab)
Reason:
     I often mixed use several editors before,and some doesnot support tanslate tab to space,some support.so tab and space mixed,this is error in python, if we can see whitespace it will help us to debug easily.

aivarannamaa, 2019-08-25:
I'm afraid Tkinter Text widget doesn't have a cross-platform way for displaying extra symbols on top of or below characters.

What about adding special background color to tabs (unless they are inside a string literal) and different background color to trailing spaces?

Dleta, 2019-08-26:
I'm a little puzzled.
Is it easy to conclude the background color meaning, which is tab and which is space or other meanings?
And,I am afraid can't see the number of whitespace characters directly.

aivarannamaa, 2019-08-29:
> Is it easy to conclude the background color meaning, which is tab and which is space or other meanings?

Nope. I was thinking of highlighting only those symbols which should be replaced or deleted. I guess the user will need extra guidance to understand this.

Dleta, 2019-08-29:
Understand,Maybe it's enough to debug.
　☆  *　.  　☆
　　. ∧＿∧　∩　* ☆
*  ☆ ( ・∀・)/ .
　.  ⊂　　 ノ* ☆
☆ * (つ ノ  .☆
　　 (ノ

aivarannamaa, 2020-12-12:
Tabs are now highlighted. Add non-breaking space etc.

thisBetter, 2020-12-13:
thanks for your work.