thonny/thonny #871: Thonny manage packages not working on Mac - no error messages given

benappio, 2019-08-30:
I can download and open Thonny normally. But then when I click Tools > Manage Packages or Thonny > Preferences, Thonny goes to a white screen with a stripe in the middle then shuts down and takes me to the Mac login page. Any help would be appreciated.

aivarannamaa, 2019-08-31:
Which macOS version are you using?

Do you mean these actions cause something like this:
[image]

or something like this: 
[image]
?

benappio, 2019-09-01:
The second picture

aivarannamaa, 2019-09-01:
Which macOS version and which Thonny version are you using?

benappio, 2019-09-01:
macOs Mojave Version 10.14.6
Thonny 3.1.2

aivarannamaa, 2019-09-03:
Unfortunately I can't reproduce this and I don't have any ideas what can cause this.
Please check, whether version 3.2.0 works better for you: https://github.com/thonny/thonny/releases/download/v3.2.0/thonny-3.2.0.dmg

idominiki, 2020-11-16:
Hello together,

I have the same problem under macOS 10.15.7. After I choose the language the App quits. It can't be open. 
I have this behavior with a Standard Users, without admin rights.

If I run the App under an Admin Account, the App works well.

I distributed the Package thru MDM Jamf Pro.

Would be great If there is a fast solution.

aivarannamaa, 2020-11-16:
@idominiki , yesterday I released version 3.3.0. Please check whether this works better: https://github.com/thonny/thonny/releases/download/v3.3.0/thonny-3.3.0.pkg

idominiki, 2020-11-16:
Unfortunately, no

aivarannamaa, 2020-11-16:
It's a pity. For now my hope is on updated Tk/Tkinter which is supposed to be published with next alpha of Python 3.10. I hope I can extract it from there and include with Python 3.9 in Thonny 4.0.

aivarannamaa, 2020-11-16:
@idominiki , one more thing -- what do you have in your macOS settings under "Dock => Prefer tabs when opening documents" ?

If it's "Always", then try setting it to "Manually".

idominiki, 2020-12-18:
Sorry for my late response. I set this from only in fullscreen to manually, but nothing changed.

Also didn't worked with the XXL version from Thonny.

aivarannamaa, 2020-12-18:
@idominiki, could you please try one more thing? I created an alternative bundle for macOS, which comes with Python 3.9 with new Tk: https://github.com/thonny/thonny/releases/download/v3.3.1/thonny-3.3.1_alt.pkg

idominiki, 2020-12-18:
Not working :(

aivarannamaa, 2020-12-18:
It's a pity. My hope is now on the next Tk release (8.6.11), which should be out in couple of weeks.

idominiki, 2021-01-07:
Hey @aivarannamaa,
just updated to Thonny 3.3.2: nothing changed.

Greets
Dominik

aivarannamaa, 2021-01-07:
This was expected, as I didn't upgrade Tk. In fact, even official Python distribution doesn't have Tk 8.6.11 yet.

aivarannamaa, 2021-02-27:
Please try whether https://github.com/thonny/thonny/releases/download/v3.3.5/thonny-3.3.5_alt.pkg works better for you. It's with an experimental Universal 2 variant of Python 3.9.2 with Tk 8.6.11.

idominiki, 2021-03-04:
> Please try whether https://github.com/thonny/thonny/releases/download/v3.3.5/thonny-3.3.5_alt.pkg works better for you. It's with an experimental Universal 2 variant of Python 3.9.2 with Tk 8.6.11.

Just updated to Thonny 3.3.5_alt: Same problems here

aivarannamaa, 2023-08-19:
It's worth trying the newest version with Tk 8.6.13: https://github.com/thonny/thonny/releases/tag/v4.1.2