thonny/thonny #873: Thonny packages issue on Raspberry Pi 3/ Raspbian Desktop

BioHazard1282, 2019-09-01:
Hello, I am trying to use Thonny for an assignment for a class. I cannot install it on Raspbian because it gives me the following errors:

`Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/__init__.py", line 65, in launch
    bench = workbench.Workbench(server_socket)
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 151, in __init__
    self._init_menu()
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 317, in _init_menu
    self.get_menu("file", "File")
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 1281, in get_menu
    menu = tk.Menu(self._menubar, **get_style_configuration("Menu"))
  File "/usr/lib/python3.5/tkinter/__init__.py", line 2720, in __init__
    Widget.__init__(self, master, 'menu', cnf, kw)
  File "/usr/lib/python3.5/tkinter/__init__.py", line 2151, in __init__
    (widgetName, self._w) + extra + self._options(cnf))
_tkinter.TclError: invalid color name "#ffff"

`

Thank you for your help.

aivarannamaa, 2019-09-01:
Have you changed your Raspberry theme configuration files by hand?

Can you post me your `/etc/xdg/lxsession/$DESKTOP_SESSION/desktop.conf` and `.config/lxsession/$DESKTOP_SESSION/desktop.conf` ?

BioHazard1282, 2019-09-01:
I cannot post any of those because none of those directories are valid.

 
> Have you changed your Raspberry theme configuration files by hand?

I do not know how to...

aivarannamaa, 2019-09-01:
Which Raspbian version are you using? Stretch (older one) or Buster (came out this summer)? Have you changed your desktop environment? What does `echo $DESKTOP_SESSION` give you in Terminal?

BioHazard1282, 2019-09-01:
The $DESKTOP_SESSION Gives me this: **LXDE-pi**

The current version of Raspbian is this:

pi@Bio_Hazard1282_rPi3:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

closed 2019-10-11 by aivarannamaa