thonny/thonny #878: RTL text like (Persian/Arabic , ...) issue

Farshidmi, 2019-09-01:
Hi
Thanks for your great job.
Thonny dosen't support RTL language :
When we type some thing Thonny change orders and put word LTR.
And the out put actually reorder character.

Regards

[image]

aivarannamaa, 2019-09-02:
Can you please check whether inserting RTL text works in IDLE.

A link for myself: https://stackoverflow.com/questions/20306726/right-to-left-text-in-tkinter

Farshidmi, 2019-09-02:
Thanks for your reply.
In IDLE every things is OK when I type only Farsi(Persian) words . 
But in complex script ( Persian words with English) it does not OK **but** this happens in many editors and we used to dealing this issue :) 

[image]

Farshidmi, 2020-12-15:
Hi 
I want to report In this version the output is OK and shows correctly Persian phrase. 
1) 
Same as previous version, the phrase reordered in the editor (Like English it show the phrase LTR instead of RTL)
2)
The editor/and output dose not support "farsi ya" and change it to "Arabid ya".
this is Farsi ya: ی
this is Arabic ya: ي

Thanks,

[image]

aivarannamaa, 2020-12-16:
Thank you for the feedback! Thonny 4.0 will switch to newer Tk version, I'll hope this will help me fix this.