thonny/thonny #880: suggestion: toolbar

pdalet, 2019-09-03:
Can you increase the size of the toolbar ?

Thanks a lot (see mu-editor)

Philippe

aivarannamaa, 2019-09-03:
Do you mean increasing size of the toolbar icons?

pdalet, 2019-09-03:
Le Tue, 03 Sep 2019 03:18:45 -0700,
Aivar Annamaa <notifications@github.com> a écrit :

> Do you mean increasing size of the toolbar icons?
> 

Yes

Philippe

aivarannamaa, 2022-02-09:
In Thonny 4.0 the size of the toolbar icons will be dependent on the pixel size of default font. This means on large monitors they most likely will be bigger.

closed 2022-02-09 by aivarannamaa

lylewh, 2024-12-30:
Hi all,
        I found the required settings to change the toolbar and button size in Thonny.
Please see screen shots
They come under Tools\Options\General
[image]
[image]
Makes using the program far more comfortable
Regards
Lyle