thonny/thonny #882: hi , i can't install dlib

giosin, 2019-09-04:
i have installed dlib (dlib-18.17.100-cp34-none-win32.whl) on my windows 32 bit python 3.4 pc, but i can't use it  in thonny ...

i tried every way throught manage packages.

any help?

aivarannamaa, 2019-09-04:
How do you know you can't use it in Thonny? Which Thonny version?

giosin, 2019-09-04:
thonny 3.1.2, when i try in python idle it works, 

in thonny :

> module = self._original_import(*args, **kw)
> 
> ModuleNotFoundError: No module named 'dlib'

the only version i managed to install is 18.17.100 (with the current fault), i tried the latest version because i see it runs in python 3.7, no success

aivarannamaa, 2019-09-04:
`cp34` in `dlib-18.17.100-cp34-none-win32.whl` means it's compiled for Python 3.4, but Thonny 3.1 doesn't include nor support Python 3.4.

giosin, 2019-09-04:
but i can use numpy , opencv and other libraries?

i should unistall thonny and i install an earlier version?

aivarannamaa, 2019-09-04:
> but i can use numpy , opencv and other libraries?

yes, the are published for all recent Python versions

> i should unistall thonny and i install an earlier version?

It looks like there is 64-bit dlib for Python 3.6 (https://pypi.org/simple/dlib/). I recommend you install separate 64-bit Python 3.6, install dlib there (https://pysource.com/2019/03/20/how-to-install-dlib-for-python-3-on-windows/) and let Thonny use this Python as back-end (Tools => Options => Interpreter)

giosin, 2019-09-04:
unistall previous version, install 3.6.8, Tools => Options => Interpreter, same as machine,
the first attempt was a heavy crash of windows (i don't know why), after that everything is perfect.

as a begginer programmer i have to say , that i tried a lot IDE , thonny is the best for me.
Thanks a lot

aivarannamaa, 2019-09-06:
I'm glad you got it working!

closed 2019-09-06 by aivarannamaa