thonny/thonny #886: Cmd-backtick and others crash Thonny

juanfal, 2019-09-06:
It has been happening since first Thonny version I've used.  It seems to be connected more with TkInter…   When you press ⌘-` (and other combinations using Cmd on Macs) it crashes without a simple warning.  Total crash.  This is quite annoying for students used to Windows who don't know the meaning of the basic Command key on Macs and try to write [ wrongly using the ⌘  instead the ⌥  key

aivarannamaa, 2019-09-08:
Thanks for letting me know! I'll try to reproduce it when I get around my Mac. Which keyboard layout are you using?

juanfal, 2019-09-09:
Thanks Aivar

Ours is Spanish

fstengel, 2019-09-15:
Same thing happens using a French keyboard. The culprit seems to be an uncaught exception. See the exerpt of the crash report:
```
Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[TKMenu submenuAction:]: unrecognized selector sent to instance 0x7fac6049e1b0'
abort() called
terminating with uncaught exception of type NSException
```
I can send the whole crash report, if you need it...

aivarannamaa, 2019-10-03:
I could reproduce this, but it looks like the solution is out of my reach: https://bugs.python.org/issue34455

I created a Tk ticket
https://core.tcl-lang.org/tk/tktview/5849df7852c48cb763b11a11d848a68482b0f828 and I'm afraid I can't do more. When it get fixed, it will come to Thonny when Python includes new Tk version

closed 2019-10-03 by aivarannamaa