thonny/thonny #887: Thonny interpreter crashs

Skiski, 2019-09-06:
Hi,
In my school I've tried using thonny that I installed last Year but it's not working very reliably. We are using windows 7 and it was working before the sumer hollydays. I'm mainly using Thonny 3.1.2. But I have the same problem with 3.2 beta or 3.0.8.

When I launch Thonny, most of the time, the buttons on the top of the windows (run, debug, load, save...) are not showing up and after some times, winsows says that Thonny is not responding. Or something the button shows up, but the interpreter does not start. The bottom window stays empty. One way to solve both problems is to kill the python process and the Thonny works.

And sometimes it crashes when I try to run a file. Once again, killing the python process "solves" the problem, or at least "unfreeze" Thonny.

I've noticed that when I run a file, a second python process is created then disappear after a few seconds. I don't know if it is a normal behaviour.

I've had some similar problems befor the summer, when I upgraded to 3.1.2, but it was way less often. And I had no problems with 3.0.8, but now even this version have similar problems.

I've always done fresh installs for newer versions. The only thing I've done is creating a custome.py file :
```python
import os
import thonny

dossierperso=os.path.join(os.environ["LOGONSERVER"],"professeurs",os.environ["USERNAME"],"Mes documents")
dossierpersoeleve=os.path.join(os.environ["LOGONSERVER"],"eleves",os.environ["USERNAME"],"Mes documents")
if os.path.isdir(dossierperso):
    os.environ["THONNY_USER_DIR"]=os.path.join(dossierperso,".thonny")
elif os.path.isdir(dossierpersoeleve):
    os.environ["THONNY_USER_DIR"]=os.path.join(dossierpersoeleve,".thonny")

thonny.THONNY_USER_DIR = os.environ["THONNY_USER_DIR"]
```

I've also installed a few pacakges using this script :
```python
import pip._internal as pip
packages=["pygame","matplotlib<3.1.0","sympy","openpyxl","win32com"]
pip.main(["install","--no-color"]+packages+["--upgrade"])
```
I don't know if I can send some kind of log file to help.

Romain Janvier

aivarannamaa, 2019-09-07:
Hi!

Invoking pip in customization script is not good idea -- I believe this is what makes your start-ups slow. Better install required packages into bundled Python before redistributing Thonny (or after installing shared Thonny).

BTW. In 3.2.1 (uploaded yesterday) I fixed an issue that may be relevant in classroom setting: https://github.com/thonny/thonny/issues/854

aivarannamaa, 2019-09-07:
If you tell me more about your requirements / classroom specifics, then maybe I'm able to propose another solution.

Skiski, 2019-09-07:
It's not a problem of "slow" start-up, but rather taht it is not starting correctly. When it's working, it just takes a few seconds to launch.
I'll try 3.2.1. A technician has made a new disk image under windows 10 and he took my version of Thonny 3.1.2, and from what I've seen it's working great. But this image will only be deployed on a few computers and not the one I use with my students.
To install packages into python, I'm a bit stuck since opening windows console is blocked and I don't really have another choice than installing directy from inside Thonny. But I worked great until now.

What I really need is to choose where the user data are located (which is done with customize.py and I need to be able to add new packages when I need them or when I install a new version of Thonny.

Maybe I could try to do a .bat script that launch the correct python executable. We already have python installed on computers but it does not have the packages I need (pygame, matplolib,...). I don't have administrator rights and I'm pretty sure that anything I could do install on this version of Python would be erase rather quickly. 

By the way, I've tried using this version of Python as Thonny's interpreter and I have the same problem of it hanging. I've also tried the option inside thonny to create a new python environment but it is also hanging...

I'll try monday to see if 3.2.1 solves the problem but I'm pretty sure it's not that since I've used thonny for 2 years now in our multi-users system and it has always worked great. I'll also try if I can install things directly into thonny's python. I'm a linux user, so I'm not very efficient under windows, and all the restrictions that we have don't really help.

Thanks for your help and your answers.

Romain Janvier

aivarannamaa, 2019-09-08:
Can you somehow use the portable variant of Thonny? (I forgot to upload it for 3.2.1 (will do tomorrow) but here is https://github.com/thonny/thonny/releases/download/v3.2.0/thonny-3.2.0-windows-portable.zip). You could install required packages to it (via command line, because "Manage packages" dialogs does `pip install --user`) zip again and distribute to your students.

Or, if you want to streamline students installing packages themselves, you could prepare a `requirements.txt` file and have them install it via `Manage packages` dialog (ability to install via requirements file was introduced in 3.2.0)

I don't understand, why your classroom admin can't install a few extra packages to shared Thonny after he installs Thonny itself. If you now Thonny's install location, then you could prepare a bat-file for him, eg:

```
"C:\Program files (x86)\Thonny\python.exe -m pip install matplotlib<3.1.0 pygame"
```

Skiski, 2019-09-08:
I'll give a try to the portable version, even an older one.
And for the admin, he is not in my school and he comes at the beginning of each year and once the disk image is done, it is done. That's why I prefer to use "my" version of Thonny that I can modify or update as I want.
I neither want to distribute Thonny to each student, because it would occupy a lot of disk space for nothing.
I put Thonny on a shared disk and I distribute the link to the executable to my students. This way, there is only a version of Thonny installed and I can update or modify it as I want.

aivarannamaa, 2019-09-08:
Sorry, I now noticed that I misread your initial comment -- I thought you were doing `pip.main(...)` in the customization script. 

> I don't know if I can send some kind of log file to help.

There are a bunch of log files in THONNY_USER_DIR. These can be helpful. NB! They get overwritten each time Thonny is launched, so you need to copy them right after unsuccessful session, before launching Thonny again.

Skiski, 2019-09-08:
Ok, so you mean that the way I install packages should be the problem, even if there are more elegant ways to do it?
I'll try to collect the log files and attach them.

aivarannamaa, 2019-09-09:
> Ok, so you mean that the way I install packages should be the problem, even if there are more elegant ways to do it?

No, I don't see why installing via (unofficial) pip API should be the problem. Currently I don't have any ideas what causes this.

Skiski, 2019-09-09:
Ok, I've tried, and the only log file with something is frontend.log with:
```
INFO: Could not allocate console. Error code: 31
ERROR: Problem with finalizing console allocation
Traceback (most recent call last):
  File "V:\Thonny\lib\site-packages\thonny\running.py", line 525, in _check_alloc_console
    child.stdin.flush()
OSError: [Errno 22] Invalid argument
```
I had to kill the python process because the thonny was frozen just before launching the shell. I don't know if this log was written before or after the kill.

aivarannamaa, 2019-09-09:
This may be the key. You mentioned that in your setting opening windows console is blocked -- this may cause the error above.

I'll consider making this part of the code optional. If you switch it off, you'll lose the ability to interrupt your programs with Ctrl+C (but you still can restart the back-end with "Stop/Restart"), but hopefully you will gain better stability.

Skiski, 2019-09-09:
It is strange because it works sometimes, and sometimes not. And When it fails, I've got a python process staying, even if I quit python.
It even happened when I installed python 3.2.1. It hanged during the compilation of libraries...
Can I try to modify directly the file to test if it is better?

aivarannamaa, 2019-09-09:
You could try commenting out this line: https://github.com/thonny/thonny/blob/master/thonny/running.py#L86

Hopefully it doesn't break anything else.

Skiski, 2019-09-09:
It seems to work, but it opens an empty terminal window.

aivarannamaa, 2019-09-09:
In the same module, there are several occurrences of `.replace("thonny.exe", "python.exe")` -- try deleting them. The idea is to use `pythonw.exe` for back-end instead of `python.exe`.

If it doesn't work then I'll take time to do it properly

Skiski, 2019-09-09:
I've removed the 2 or 3 occurences of .replace("thonny.exe", "python.exe") but I still have the windows opening.

aivarannamaa, 2019-09-09:
I'll try to find time for looking into this tonight.

Skiski, 2019-09-09:
Ok, thanks.

Skiski, 2019-09-09:
Ok, I've realised that when I change things in running.py, I was using another python as interpreter. I've switched to the included one, and now I've got this message:
```
Delegated to an existing Thonny instance. Exiting now.
Backend terminated (returncode: 1)
Use 'Stop/Restart' to restart the backend ...
```
It happens each time it use stop.

aivarannamaa, 2019-09-11:
I think I got it working when console allocation fails. You are in the fallback mode when `sys.executable` ends with `pythow.exe` (instead of `python.exe`).

If you try to interrupt the program in fallback mode, you should get error message, which recommends Stop/restart

aivarannamaa, 2019-09-11:
Most likely I won't release next version until October, but you can overwrite `thonny` package in your 3.2.1 installation with current master. (Actually, you can test the core of this fix by updating just `thonny.running`).

Skiski, 2019-09-11:
Ok, I'll try it tomorrow and let you know. Thanks a lot.

Skiski, 2019-09-14:
I've tried with a portable version of 3.2.1, with absolutely no modifications (no new package installed, no customize.py). I've replaced all the files in site-package/thonny and it's still the same.
I've still got the same "freezes".
I've also tried on one of the new computers with windows 10 and even the base 3.2.1 version is working great without any problem.
It seems to be a windows 7 related problem, since the restrictions on the console are the same on the computers with windows 10.
I have no idea on how I can provide more information to help you.

aivarannamaa, 2019-09-14:
Can you check whether the fallback mode kicks in (`sys.executable` ending with `pythow.exe`)?

Skiski, 2019-09-14:
Ok, I'll look at it monday. How can I check that? In the task manager? I've made a script that kills all python.exe processes, and it still works with this version, so I'm pretty sure that it is only python.exe that is used.
Since I've only change the files in the thonny library, did it change what happens with thonny.exe? Does it just launch what is in the thonny library or are there some part that are precompiled, and thus, still causing the problem?

aivarannamaa, 2019-09-14:
> How can I check that?

```
import sys
print(sys.executable)
```

> Since I've only change the files in the thonny library, did it change what happens with thonny.exe? Does it just launch what is in the thonny library or are there some part that are precompiled, and thus, still causing the problem?

thonny.exe should take changed py files into account at next run. At least that's how I debug problems which are not reproducible in my dev environment.

Skiski, 2019-09-16:
I've just checked and it is calling python.exe.

aivarannamaa, 2019-09-16:
Ouch, my bad -- I assumed `_check_alloc_console` raised exception in your environment, but now I noticed that the exception was captured and only logged. This means the fallback does not kick in.

Can you please check whether adding `raise` after this line helps: https://github.com/thonny/thonny/blob/master/thonny/running.py#L607

aivarannamaa, 2019-09-16:
To be sure, you can also add a `raise RuntimeException("...")` after this line: https://github.com/thonny/thonny/blob/master/thonny/running.py#L599

Skiski, 2019-09-16:
Whenever I add any of the two, I get an error saying " 'NoneType' object has no attribute 'event_generate' ".
Just to be sure, I've got:
```
          if not result:
                err = ctypes.get_last_error()
                logging.info("Could not allocate console. Error code: " + str(err))
                raise RuntimeException("...")
```
and
```
           except Exception:
                # May happen eg. when installation path has "&" in it
                # See https://bitbucket.org/plas/thonny/issues/508/cant-allocate-windows-console-when
                # Without flush the console window becomes visible, but Thonny can be still used
                logging.getLogger("thonny").exception("Problem with finalizing console allocation")
                raise
```

aivarannamaa, 2019-09-17:
Unfortunately I can't reproduce this. Can you please send me the whole stacktrace?

Skiski, 2019-09-17:
I'm sorry, but I have no idea on how to do that. On my linux computer, I think I could do it, but on windows, I have no idea. Should I make a batch script or something like that? Or is there just a file that I can find somewhere?

aivarannamaa, 2019-09-18:
> I get an error saying " 'NoneType' object has no attribute 'event_generate' ".

Did you get only this line without traceback? In regular message box (instead of Thonny's message box with separate text widget for error)?

> Or is there just a file that I can find somewhere?

In Thonny's user directory there should be some log-files (overwritten on each session), which can contain more info.

Skiski, 2019-09-18:
I've got another window with only this error message. And the log files are empty.

I'll try to run thonny from command line to see if I've got more information.

aivarannamaa, 2019-09-18:
> I'll try to run thonny from command line to see if I've got more information.

I'm afraid you won't, because thonny.exe doesn't have a console.

Skiski, 2019-09-18:
Can't I run "python.exe some_python_file" to launch thonny ? I've seen that it's what the linux script is doing, so I assume, I can try to do the same with the windows version.

aivarannamaa, 2019-09-18:
yes, this should be possible

Skiski, 2019-09-18:
I can't do that today, but I'll try it this week. Thanks for your help.

Skiski, 2019-09-24:
I was able to run Thonny from command line and when python hanged, I had no output, but when I use CTRL+C on the terminal, I've got this message:
```
Traceceback (most recent call last):
   File "U:\romain.janvier\Mes documents\thonny-3.2.1-windows-portable\lib\runpy.py", line 193, in _run_module_as_main
   "__main__", mod_spec)
   File "U:\romain.janvier\Mes documents\thonny-3.2.1-windows-portable\lib\runpy.py", line 85, in _run_code
   exec(code, run_globals)
   File "U:\romain.janvier\Mes documents\thonny-3.2.1-windows-portable\lib\site-packages\thonny\__main__.py", line 3, in <module>
   launch()
   File "U:\romain.janvier\Mes documents\thonny-3.2.1-windows-portable\lib\site-packages\thonny\__init__.py", line 166, in launch
   bench.mainloop()
   File "U:\romain.janvier\Mes documents\thonny-3.2.1-windows-portable\lib\tkinter\__init__.py", line 1283, in mainloop
   self.tk.mainloop(n)
   File "U:\romain.janvier\Mes documents\thonny-3.2.1-windows-portable\lib\tkinter\__init__.py", line 1700, in __call__
   def __call__(self, *args): 
   KeyboardInterrupt
```
I don't know if this can help.

Skiski, 2019-09-25:
Another teacher confirmed me that he has the same problem in his school, which is also under windows 7. For the moment I'm using 2.1.11 with my students because it is already installed in the computers (I was using a more recent version that was on a shared disk) and they have no problems. I'be tested using this version of Python as interpreter for the most recent Thonny, but it is the same.

Have you changed the way you launch the python interpreter between version 2 and 3 of Thonny?

aivarannamaa, 2019-09-27:
Does this other teacher have similar technical restrictions (blocked console)?

Does Thonny 3 work for you when you don't user your customization?

> Have you changed the way you launch the python interpreter between version 2 and 3 of Thonny?

There were [lot of changes](https://github.com/thonny/thonny/blob/master/CHANGELOG.rst#300b1-2018-08-11) between version 2 and 3, but I don't recall anything to suspect.

One important thing, that was changed was bundled Python and Tk versions.

I'll try to find time for preparing a Windows 7 VM and trying to reproduce the problem.

> When I launch Thonny, most of the time, the buttons on the top of the windows (run, debug, load, save...) are not showing up

Is the window completely empty? I occasionally see this while developing Thonny, but I thought it is a problem of my development environment.

Skiski, 2019-09-30:
The other teacher have a similar system, with the same restrictions on Windows 7.

I've tried with various versions of thonny with and without any modifications and it is the same. I even have tried with other versions of python and it is the same.

When python hangs at launch, I have the main window, with the top menu bar (file, edit...) and the two parts : the text editor and the interpreter. But they are empty, without anything and I d'ont have the icons at the top of the text editor (new file, open, save, run, stop...).

Skiski, 2019-10-03:
I've trued 2.1.22 and it works great. I've installed everythingI needed, made the modifications tu use the same version of python than Thonny and I had absolutly no problem. I've tested it with my students and everything worked. I'll try the first versions of 3.x to see if the problems occurs between 2.1.22 and 3.0 or between more recent versions.

aivarannamaa, 2021-01-19:
Closing, because Thonny is nearing the end of supporting Windows 7

commit c936bd81 (2019-09-11): FIXED #887: Fallback to using pythonw.exe when console allocation fails
closed 2021-01-19 by aivarannamaa