thonny/thonny #888: Thonny-xxl-3.2.0 crashes

DLatreyte, 2019-09-06:
I've just tried to test Thonny-xxl-3.2.0 on several Mac and, at the first launch I only have this message (translation from french) : "Thonny.app is damaged and cannot be opened. You should put it in the trash".
I've encountered no problem with Thonny-3.2.0.
Bye

aivarannamaa, 2019-09-06:
Please try 3.2.1: https://github.com/thonny/thonny/releases/tag/v3.2.1

DLatreyte, 2019-09-09:
Hi,

it works! Congratulations!

Best regards.
————
David Latreyte
david.latreyte@gmail.com

fstengel, 2019-09-13:
I have the same issue and it has not been solved with xxl-3.2.1 I tried on two machines running 10.13.6

aivarannamaa, 2019-09-14:
@fstengel, does regular variant (without xxl) of 3.2.1 work for you?

fstengel, 2019-09-14:
The regular variant of 3.2.1 starts without problems...

fstengel, 2019-09-15:
This is visibly a gatekeeper issue. If I start thonny-xxl for the first time having disabled gatekeeper ("install from anywhere" in the Security preference pane) everything works well.

davidalexisnyt, 2019-09-17:
For me, nothing past v 3.1.2 works on either Mac or Windows 10.  On Windows, the IDE hangs when it tries to interact with a device - file view shows nothing from the device, attempts to run code hangs, the display of the local file list is corrupted.
On a Mac, the device file list displays, but any attempt to load interact with the device file list causes the machine to crash and reboot.

aivarannamaa, 2019-09-17:
@davidalexisnyt, which device are you using?

In which way is the local files list corrupted?

davidalexisnyt, 2019-09-17:
It's like a display issue - like the line height was about half the height of the font being used, so everything was just squished.

aivarannamaa, 2019-09-17:
@davidalexisnyt, what is the resolution of your display? Have you changed any scaling settings under "Tools => Options => General"?

davidalexisnyt, 2019-09-18:
I had not changed any scaling settings before, but I tried various options last night.  Setting it to 2.0 or under seems to get rid of the overlapping lines.
My Windows machine is a Microsoft Surface Book, with a native resolution of 3000 x 2000, and the display scaling factor set to 200% (which I assume would make it the resolution behave like 1500x1000.  Thonny 3.1.2 seems to honor the display settings, but 3.2 seems like it gets confused and tries to use the native resolution. 
Also, in the Files pane, the device files panel is empty, and (with the portable version) got a dialog showing errors in backend.py.  I guess it's having issues communicating with the device.  The REPL works, though.

aivarannamaa, 2019-09-18:
@davidalexisnyt 

> I guess it's having issues communicating with the device.

Which device are you using?

davidalexisnyt, 2019-09-18:
Generic NodeMCU-type ESP8266 board.  It works well with Thonny 3.1.2.

njkavon, 2019-09-23:
Also can't run xxl Thonny 3.1.2. on macOS Catalina Beta 6. 
Prints app is corrupted - move to trash.

Regular one prints that app can't be checked for malicious activity - contact developer, but just turning off gatekeeper makes it work.

Would be glad to help with testing xxl version.

njkavon, 2019-10-02:
Both packages runs great, with first run notice window as usual. Good job!

Noticed greyed out menu items, different for each package(regular, xxl respectively). 
Nothing serious, because preferences and exit is functional through File menu item.

[image]
[image]

aivarannamaa, 2019-10-02:
> Noticed greyed out menu items, different for each package(regular, xxl respectively).

Thanks for letting me know! I'll investigate this

aivarannamaa, 2019-10-11:
Greyed out menu items were solved under #905

closed 2019-10-11 by aivarannamaa