thonny/thonny #890: Error when installing Thonny 3.2.1

fa201, 2019-09-07:
I download v3.2.1 (last stable release) and tried to install on Xubuntu 18.04. Note: one month ago I succeeded to install.

In packaging directory, I launched:
`sudo bash install.sh /opt`
I got the following error:
```
Copying files to /opt/thonny ......................................... 
/opt/thonny already exists. I need to clear it. Is it OK? [Y/n]: Y
[Errno 2] No such file or directory: '/opt/thonny/templates'
```
Is there anything I could do with this issue ? Thanks.

aivarannamaa, 2019-09-07:
I suggest you delete old directory manually (`sudo rm -rf /opt/thonny`) before installing Thonny.

fa201, 2019-09-07:
It gives the same message.

fa201, 2019-09-07:
I have the same error with 3.2.0 and 3.2.0.rc1.

fa201, 2019-09-07:
I did not download the right file (x86_64). I close the issue. Apologizes for the inconvenience...

closed 2019-09-07 by fa201