thonny/thonny #906: Suggestion: line wrap in editor

campbelljc, 2019-09-16:
Hi,

I am using Thonny for my intro programming class. When I show a Thonny window on the projector screen, I have to increase the font size so that everyone can see the code. However, this means that my lines have to be short, otherwise, people in class won't be able to see the ends of long lines since they are cut off. Would it be possible to introduce some kind of line wrapping in the editor window, i.e., when a line of code reaches the end of the window, it will continue on the next line?

Thanks for the program by the way, very useful!

aivarannamaa, 2019-09-16:
IMO it would be good idea with free-form languages, but wouldn't it cause confusion with Python?

When I was teaching I did write short lines.

Ghanisen, 2020-03-14:
Hi @aivarannamaa,
@campbelljc's remark makes sense and I would like to see **line wrapping** introduced **as an option** in the editor for code and comments. This way, those who don't need it won't use it.

By the way, I tought that the 'maximum line length' in Tools > Edito allows to set up wrapping after reaching this length, but I set it to 90 and nothing changed in my editor?

Thumbs up for your very handy python IDE!

aivarannamaa, 2020-03-14:
I'll see whether I can find a way to make soft-wrapped lines visually distinguished from hard-wrapped lines.

carloswm85, 2020-05-01:
Major editors and IDE allow line-wrapping. It's a must feature for all of them. Thonny should't be an exception.

cybarspace, 2021-01-16:
> Major editors and IDE allow line-wrapping. It's a must feature for all of them.

Thonny doesn't aim to be a general-purpose Python editor/IDE though. For absolute beginners, line wrapping would cause confusion. You could teach your students to use the Home/End keys, and the [Black extension](https://github.com/Franccisco/thonny-black-code-format) for Thonny.

r0the, 2021-01-16:
As a workaround you could use [Python line joining](https://docs.python.org/3/reference/lexical_analysis.html#explicit-line-joining), e.g.

``` python
if font_is_big and \
  line_is_long:
    use_line_joining(
      "explicit",
      "implicit"
    )
```

aivarannamaa, 2021-01-19:
Closing for now. Please reopen, if you have an idea how to solve it with Tkinter

closed 2021-01-19 by aivarannamaa

LisaBBC, 2022-06-21:
It's fine to leave wrapping off by default but still have it as an option, maybe in UI expert mode. It's such a lovely lightweight handy IDE.

I am an absolute beginner and doing a class but the instructions of the lessons are in long blocks of comment and hard to read with thonny. It's really sad, because I love this Thonny but for this.

MattTheBanana, 2022-12-14:
Hi,
I think soft-wrap is generally a nice feature because it allows user to change their window size without having to scroll to see information, which is it's purpose. Just a little option somewhere would be very useful in my opinion.