thonny/thonny #907: Run current line or selected text

mickley, 2019-09-17:
Currently, one can only run an entire file at once with F5.  It would be very useful to run sections, of a file, either the current line, or selected text, or both. 

In the case of the current line, after running, advancing the cursor down a line is useful as well.

aivarannamaa, 2019-09-19:
I have considered adding support for executing code cells (https://docs.spyder-ide.org/editor.html#defining-code-cells), but it hasn't been a priority. I hope this would cover your needs as well.

mickley, 2019-09-19:
@aivarannamaa code cells would allow some more flexibility, but just running selected text would be simpler, without adding stuff to a file.

Is there a way to get the text of the selected line in the editor, or a way to get the highlighted text in the editor? If so, maybe I can hack something myself when I have some time, just pass it to RUN -c

aivarannamaa, 2019-09-19:
Maybe we could use same command for both -- running current cell when nothing is selected and running selection otherwise?

Actually, there is related work already in the repo: https://github.com/thonny/thonny/blob/master/thonny/plugins/cells.py

I haven't activated it because I didn't want to crowd menus with non-essential commands. But now it occurred to me that I could activate it only in expert mode (Options => General)...

mickley, 2019-09-19:
@aivarannamaa Yes, that is what some other editors do (e.g., RStudio for R).  Thank you!

juanfal, 2019-09-20:
It could be easier, IMHO, to send the line to the shell and execute it there, kind of a macro

aivarannamaa, 2019-09-24:
Duplicate of #636

closed 2019-09-24 by aivarannamaa

bquast, 2022-10-01:
Now that 4.0 is out, I would revisit this again.

For an IDE focused on learning, users need to run code line-by-line in order to build understanding in manageable chunks.

aivarannamaa, 2022-10-02:
It's still in the TODO list, as #636.

Until then the users can use debugger for running the code statement-by-statement.

bquast, 2022-10-02:
Hello, thank you. I was not aware of this option.

That solves my problem. But a beginning might still download a (python2) script from the internet, and want to start running parts. If the script ends with printing some results (without brackets) it is still difficult

aivarannamaa, 2022-10-02:
Luckily the recent Python versions handle Python 2 prints quite nicely:

```
Traceback (most recent call last):
  File "/home/dd/thm/ff.py", line 7
    print 345
    ^^^^^^^^^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(...)?
```