thonny/thonny #909: OS Error

BioHazard1282, 2019-09-18:
I keep receiving the error after using turtle graphics in the newest version of Thonny:

```
Traceback (most recent call last):
  File "E:\School\A2\Thony 3.1.1\lib\site-packages\thonny\workbench.py", line 1449, in event_generate
    handler(event)
  File "E:\School\A2\Thony 3.1.1\lib\site-packages\thonny\assistance.py", line 138, in handle_toplevel_response
    self._explain_exception(msg["user_exception"])
  File "E:\School\A2\Thony 3.1.1\lib\site-packages\thonny\assistance.py", line 178, in _explain_exception
    + _error_helper_classes["*"]
  File "E:\School\A2\Thony 3.1.1\lib\site-packages\thonny\assistance.py", line 176, in <listcomp>
    for helper_class in (
  File "E:\School\A2\Thony 3.1.1\lib\site-packages\thonny\assistance.py", line 491, in __init__
    super().__init__(error_info)
  File "E:\School\A2\Thony 3.1.1\lib\site-packages\thonny\assistance.py", line 478, in __init__
    self.last_frame_module_source = read_source(self.last_frame.filename)
  File "E:\School\A2\Thony 3.1.1\lib\site-packages\thonny\common.py", line 252, in read_source
    with tokenize.open(filename) as fp:
  File "E:\School\A2\Thony 3.1.1\lib\tokenize.py", line 447, in open
    buffer = _builtin_open(filename, 'rb')
OSError: [Errno 22] Invalid argument: '<string>'
```

aivarannamaa, 2019-09-19:
Did you get this error while running a script with F5 or after executing a command in the shell?

BioHazard1282, 2019-09-21:
Just running it normally from the shell %Run littleTurtles.py

aivarannamaa, 2019-09-22:
Does it happen for each Turtle program? Can you send me a program that causes this?

BioHazard1282, 2019-09-22:
Sure, allow me to get on my Pi when I have the time. Currently busy at
the moment

aivarannamaa, 2019-09-27:
I now noticed that according to directories in your stacktrace, you were not using last version. Please test 3.2.1, I believe this is fixed there.

BioHazard1282, 2019-09-27:
I was using the newest version? Version 3.3.1 for Windows 10

aivarannamaa, 2019-09-27:
3.3.1 doesn't exist yet. 3.2.1 is newest

BioHazard1282, 2019-09-27:
I'm running 3.1.1, my mistake

But anyways, I downloaded 3.1.1 from thonny.org

aivarannamaa, 2021-01-19:
Assuming it now works for you with latest Thonny. Please reopen if not.

closed 2021-01-19 by aivarannamaa