thonny/thonny #910: Backend Terminated

iris37, 2019-09-18:
Getting the following error message every time when I click the Stop/Restart button. My program is supposed to put esp32 in deepsleep mode every 30 seconds and then use external sources to wake it up. I have try to terminate the program when it is awake but the following error message still occur. 
[image]

aivarannamaa, 2019-09-19:
Does this error happen when you wake your device up?

Gianowo, 2019-11-07:
I have the same problem which arose spontaneously,
My source code was empty as I opened up a new file and went to the bathroom.
I came back and i recieved this error message in red
Backend terminated or disconnected. Use 'Stop/Restart' to restart ...
This same problem happened to my work computer to and reinstalling doesnt seem to solve the problem. Any tips?

aivarannamaa, 2019-11-07:
@Gianowo , were you using MicroPython back-end as well? Which Thonny version, which platform?

Gianowo, 2019-11-07:
@aivarannamaa I'm not sure if i was using MicroPython or not but it was Version 3.2.1 on WIndows 10

aivarannamaa, 2021-01-19:
Please try Thonny 3.3 and reopen if the problem is still there

closed 2021-01-19 by aivarannamaa