thonny/thonny #912: Broken code select(shift-down) upward

911992, 2019-09-20:
I'm on 3.1.2
Try this:

0. hold `shift` (initiating selection for good)
1. Press `home` (selecting from end/current to start of current line)
2. Press `up` (to select more lines upward)

And here it goes the bug! unexpected selection (it ignores step 1, for step(s) 2)

Another suggestion: please support ANSI standard terminal text coloring

Thanks.

aivarannamaa, 2019-09-21:
There is something fishy about the selection indeed. Thanks for noticing it!

ANSI colors are supported in version 3.2; https://github.com/thonny/thonny/blob/master/thonny/plugins/help/shell.rst#colored-output

aivarannamaa, 2021-01-19:
Can't reproduce in Thonny 3.3.2 on Windows

kihonkai, 2022-03-06:
I just installed Thonny 3.3.13 today on Windows, I think it is a combination of those steps with the keyboard but only after selection using the mouse before.  

I tried these steps:
1. Open Thonny
2. Open existing file, the cursor is at the end of the file
3. Don't mouse click anywhere in the opened file
4. Hold `shift` (initiating selection)
5. Press `home` (selecting from current position (end of line) to start of current line)
6. Press `up` (to select more lines upward) - ISSUE here, the selection fails to move up

If you left mouse click anywhere in the file, it then seems to work.  Right mouse click doesn't seem to work.  

Then another version is:
1. Open Thonny
2. Open existing file, the cursor is at the end of the file
3. `Left mouse click` on a line in the middle of the file, maybe anywhere but the last line
4. Press `Ctrl`+`End` to move to the end of the last line in the file
5. Hold `shift` (initiating selection)
6. Press `home` (selecting from current position (end of line) to start of current line)
7. Press `up` (to select more lines upward) - ISSUE here, the selection seems to originate from where the mouse click occurred 

And a version that works:
1. Open Thonny
2. Open existing file, the cursor is at the end of the file
3. `Left mouse click` after the last line of the file, the cursor moves to the end of the last line
5. Hold `shift` (initiating selection)
6. Press `home` (selecting from current position (end of line) to start of current line)
7. Press `up` (to select more lines upward), 

Another version that works:
1. Open Thonny
2. Open existing file, the cursor is at the end of the file
3. `Left mouse click` on a line in the middle of the file
4. Press `Ctrl`+`Home` to move to the end of the last line in the file
5. Hold `shift` (initiating selection)
6. Press `home` (selecting from current position (end of line) to end of current line)
7. Press `down` (to select more lines downward), 

It only seems to affect selections going up not down.  It also happens if you close file tabs and keep Thonny open then open the files from the Files panel.  I tested with a file with the following code:

```
print('one')
print('two')
print('three')
print('four')
print('five')
print('six')
```

I would like to try and contribute to this project though this might not be a "good first issue" to tackle, I will still try though, it looks like `tktextext.py` would be the file to start with for looking at the text editing.

aivarannamaa, 2022-03-16:
@kihonkai, thank you for these instructions and sorry for late reply.

I could reproduce this issue with the first set of instructions, both with Thonny 3.3.13 and the development version using newer Tk.

I also discovered that a minimal implementation using Tk-s Text widget does not seem to have this problem:

```python
import tkinter

root = tkinter.Tk()

text = tkinter.Text(root)
text.grid()
text.insert("1.0", "first\nsecond\nthird")
text.focus_set()
root.mainloop()
```

I suspect it has to do with Thonny's "smart home" implementation: https://github.com/thonny/thonny/blob/be076c44df413cc024bb758b71a345d71cf0315b/thonny/tktextext.py#L503
Maybe Tk doesn't like the introduction of the new mark (`"my_anchor"`)...