thonny/thonny #913: Crash Thonny with a interrump of code

RocoGamer, 2019-09-23:
Hello!

I put this code on the shell
~~~ python
while a < 10:
    print(a, end=",")
    a + 1
    print(a, end=",")
~~~
and you press Ctrl+C the thonny CRASH in Fedora

aivarannamaa, 2019-09-23:
Thank you for the report!

Can you please elaborate, how does it crash? Do you get an error message? Does the UI freeze?

RocoGamer, 2019-09-23:
The UI Freeze a second and the fedora say me the Thonny was crashed and I wait a minute and the program works fine now, I go to make more test for specify exactly the error when I have more time

aivarannamaa, 2019-09-24:
Related to #881

aivarannamaa, 2019-09-27:
I believe I was able to provide some remedy, but in your case the problem is related to Tkinter being very inefficient with long lines. If you remove end parameter from second print, then you should see much better performance. Thonny uses a workaround when long line is printed at once, but in this case it is much harder use similar workaround.

closed 2019-09-27 by aivarannamaa