thonny/thonny #920: No translation file found for domain: 'thonny'

vonij, 2019-09-26:
I changed the language in my raspberry 4B,then I can't open it. I try to replace following .py files with my old TF card,it'useless. how can I solve the problem?
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/__init__.py", line 145, in launch
    bench = workbench.Workbench(server_socket)
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 152, in __init__
    self._init_language()
  File "/usr/lib/python3/dist-packages/thonny/workbench.py", line 247, in _init_language
    language = gettext.translation("thonny", path, [language_code])
  File "/usr/lib/python3.7/gettext.py", line 539, in translation
    'No translation file found for domain', domain)
FileNotFoundError: [Errno 2] No translation file found for domain: 'thonny'

Ivoz, 2019-09-27:
In the mean time, if you are at all interested and it doesn't have a translation for your language currently (see folder [here](https://github.com/thonny/thonny/tree/master/thonny/locale)), you  would be very welcome to help by consulting https://github.com/thonny/thonny/blob/master/docs/translate.md

Ivoz, 2019-09-27:
You might be able to set your system language to English, then run Thonny, Choose Tools -> Options menu, then set the language manually to something; then set back your system language back.

aivarannamaa, 2019-09-27:
@vonij, did you change the language to Dutch? This was broken in 3.2.0. 

Please open your configuration file in a text editor (should be in `~/.config/Thonny/configuration.ini`) and delete the language line. Alternatively, delete this file altogether. Next time you'll start with default settings.

closed 2021-01-19 by aivarannamaa