thonny/thonny #926: Enhance Mac dmg with DropDMG

aivarannamaa, 2019-10-02:
https://c-command.com/dropdmg/

kcarnold, 2020-02-08:
I was able to repackage Thonny for my students using `dmgbuild` with the following command:

`dmgbuild -s settings.json "Thonny 3.2.6 (kcarnold repackage)" "Thonny 3.2.6 (kcarnold repackage)".dmg`

And `settings.json` was (based on the example):

```
{
    "title": "Thonny 3.2.6 (kcarnold repackage)",
    "background": "builtin-arrow",
    "format": "UDZO",
    "compression-level": 9,
    "window": { "position": { "x": 100, "y": 100 },
                "size": { "width": 640, "height": 280 } },
    "contents": [
        { "x": 500, "y": 120, "type": "link", "path": "/Applications" },
        { "x": 140, "y": 120, "type": "file",
          "path": "/Volumes/Thonny 3.2.6/Thonny.app" }
    ]
}
```

aivarannamaa, 2020-02-10:
Cool! Thank you for sharing! I didn't know about dmgbuild, and it looks easy enough.

I will post this in Thonny's wiki so that other teachers can also create custom bundles for their students.

I probably won't use this in Thonny, though. Because of permission problems in Catalina (#813), I decided to start distributing Thonny for Mac as an installer (pkg). I'll keep dmg for the portable variant, but this won't have link to target location (eg. Applications) on it, because it is meant to be copied to custom locations.

kcarnold, 2020-02-11:
The .pkg solution is even better, awesome!

closed 2020-05-09 by aivarannamaa