thonny/thonny #935: Improvement: Automatically install dependency tkinter if missing

lxrobotics, 2019-10-08:
Hi guys :wave:

I've just installed `thonny` via `pip3 install thonny` and have run into the following error:
```bash
$ thonny
...
  File "/home/alex/.local/lib/python3.6/site-packages/thonny/first_run.py", line 2, in <module>
    import tkinter as tk
ModuleNotFoundError: No module named 'tkinter'
```
Installing `tkinter` (`sudo apt-get install python3-tk`) solves that problem - but would it not be better to list `tkinter` as a dependency such that it is automatically installed if missing?

aivarannamaa, 2019-10-08:
Tkinter dependency is on different level than those of pip-installable packages. I'm not aware of any means for `apt-get install` -ing via `pip`.

On the other hand, if you have a Debian based distro, you can `apt-get install thonny`, which installs all requirements, including `python3-tk`.

lxrobotics, 2019-10-08:
All good and well :+1: Still it would be a good idea to mention this problem + remedy at least in the documentation, since the install description does discourage the usage of `sudo apt-get install thonny` ("may not be the latest version") but `pip3 install thonny" on a otherwise python-clean system does not work out of the box - with an error something which a python novice might find hard to debug.

aivarannamaa, 2019-10-08:
Good idea, I'll add links to more details to the download box.