thonny/thonny #936: Problem to upload .py files to some STM boards

aklein9999, 2019-10-08:
Whenever I am using Thonny 3.2.1 to upload a .py file to a Nucleo STM32F767 or a STM32F401 board I got an error message.

If I am using Micropython a Nucleo STM32L476 instead I don't have this problem.

Friends report the same problem to me.

Thank you very much for your help in advance!

Alex

Here is the error message:

>>> Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 304, in handle_command
    response = handler(cmd)
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 683, in _cmd_write_file
    self._write_file(generate_blocks(cmd["content_bytes"], BUFFER_SIZE), cmd["path"])
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1024, in _write_file
    result = self._write_file_via_serial(content_blocks, target_path, notifier)
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1095, in _write_file_via_serial
    self._execute_without_output(script)
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 400, in _execute_without_output
    out, err, value = self._execute(script, capture_output=True)
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 394, in _execute
    ok = self._connection.read(2)
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')

aivarannamaa, 2019-10-09:
Can you get to the REPL with these boards?

aklein9999, 2019-10-09:
Yes, the REPL is working well

aivarannamaa, 2019-10-09:
Please open Thonny package (Tools => Open Thonny program folder) and edit this line: https://github.com/thonny/thonny/blob/master/thonny/plugins/micropython/serial_connection.py#L51

Try smaller block_size (eg 30) and/or larger delay.

Alternatively, try smaller block size here: https://github.com/thonny/thonny/blob/master/thonny/plugins/micropython/backend.py#L1416

aivarannamaa, 2019-10-09:
If we get it working with different parameters, then I'll make them configurable.

aklein9999, 2019-10-09:
I just went to my PC. Last night I tried to reinstall Thonny. Now I face another - probably related -problem. The moment I want to access the Micropython Device reading or writing  (e.g. File open/Save as/Save copy) I get this error:

Traceback (most recent call last):
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1579, in event_generate
    handler(event)
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 867, in update_dir_data
    self.render_children_from_cache(msg["node_id"])
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 421, in render_children_from_cache
    self.update_node_data(child_id, name, children_data[name])
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 462, in update_node_data
    dtime = datetime.datetime.fromtimestamp(int(data["time"]))
OSError: [Errno 22] Invalid argument

aivarannamaa, 2019-10-09:
Your other error probably is unrelated to the first and is fixed in master (but not released yet): https://github.com/thonny/thonny/commit/4b61c8fbd2a511335081c32b55d9fa1e9c5ac6d9#diff-fcab114f4ddc600ab7e7a169cfcdbdac

aklein9999, 2019-10-09:
I applied your above fix, but the situation is still the same...

Here's the internal error message:
Traceback (most recent call last):
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 964, in dispatch_from_menu
    dispatch(None)
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 908, in dispatch
    handler()
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\code.py", line 693, in _cmd_open_file
    node = choose_node_for_file_operations(self.winfo_toplevel(), "Where to open from?")
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 1164, in choose_node_for_file_operations
    if get_runner().supports_remote_files():
AttributeError: 'Runner' object has no attribute 'supports_remote_files'

aklein9999, 2019-10-09:
Here is a screen shot, directly after I click "File-->Open" (even the selection keys where to open, filesystem or micropython device). 

[image]

aklein9999, 2019-10-09:
I wanted to say that the selection buttons where to open or save the file don't even appear.
As you can see, the REPL is working.

aivarannamaa, 2019-10-09:
How did you apply the fix? Did you copy whole file from the master? You should either apply only these specific changes in the commit or overwrite whole thonny package with the thonny directory in the master: https://github.com/thonny/thonny/archive/master.zip

aklein9999, 2019-10-09:
Yes, you were right, I copied the whole file. Now I reinstalled Thonny and applied only the changes in base_file_browser.py
I can go to the file system now.

Now back to the initial problem. 

In the file serial_connection.py I modified the two parameters individually, and now together t this:

def write(self, data, block_size=30, delay=0.05):

It takes forever now to upload a file, but still without success.
The change of the block_size has changed the error message a little bit.
This is the error message now:
>>> PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 304, in handle_command
    response = handler(cmd)
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 683, in _cmd_write_file
    self._write_file(generate_blocks(cmd["content_bytes"], BUFFER_SIZE), cmd["path"])
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1024, in _write_file
    result = self._write_file_via_serial(content_blocks, target_path, notifier)
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1095, in _write_file_via_serial
    self._execute_without_output(script)
  File "C:\Users\Alex\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 402, in _execute_without_output
    raise RuntimeError("Failed MP script: " + str(out) + "\n" + str(err))
RuntimeError: Failed MP script: 
Traceback (most recent call last):

  File "<stdin>", line 1, in <module>

  File "<stdin>", line 5, in __W

ValueError: odd-length string

aklein9999, 2019-10-09:
I also tried to modify this value in backend.py:
block = source_fp.read(32)
I tried 128, 64, 32  .... still no success, I cannot upload a file

aivarannamaa, 2019-10-09:
I assume you now tried saving a file from Thonny's editor, not upload from Thonny's file browser.

The write block size for this case is defined here: https://github.com/thonny/thonny/blob/master/thonny/plugins/micropython/backend.py#L43

Try reducing this as well

aklein9999, 2019-10-09:
The Thonny system files which I modified according to your suggestions I directly save to my PC.
The problem always occurs when I try to upload my own .py files to the Nucleo board directly from Thonny (File---> Save copy... ---> [MicroPython device]).

Reducing the value to BUFFER_SIZE = 32 also didn't help.

Maybe one more odd thing that might help:
After flashing the Nucleo board to factory defaults there is by standard an empty file "main.py" that only contains comments in the board's directory.
I always delete that one.

Every 20th time or so I succeed uploading a file without an error message.
However, I have to upload several files, and after a few failures the, any file I succeeded to upload disappears and the file "main.py" which I delteted before returns magically....

aivarannamaa, 2019-10-09:
I ordered these problematic boards to myself as well.

Did you install the USB driver by yourself (which one?) or was it installed automatically?

aklein9999, 2019-10-09:
They were installed automatically. As the REPL is working I assumed that the drivers are fine. First I wanted to do my development with a Nucleo 32F401, then I run into these uploading issues. I tried it on three different 32F401 boards. Then I switched to 32L467, this board works perfectly. However, it have to be quite fast in my application as I must read 4 digital channels at 20kHz (high resolution motor encoder pulses). As the 32L467 boards with MicroPython are a bit too slow for my purpose I tried the 32F767 board. And all the problems like for the 32F401 returned. Thonny is great and definitely the MicroPython IDE of my choice. Thank you very much for it! I hope it can eventually run with all uP supported Nucleo boards...

aklein9999, 2019-10-09:
Short update: 2 minutes ago I received a brand new Nucleo STM32F767ZI board.
Same problem. So we can say it is definitely not a hardware related issue.

aklein9999, 2019-10-09:
Thank you very much that you have purchased the problematic boards. I will continue my development on the Nucleo STM32F476RG boards as far as I can. Once you have found the problem I will move the project to the 767 board.

aivarannamaa, 2019-10-15:
@aklein9999, how did you install MicroPython to your boards? Did you use dfu files from micropython.org/downloads, built the firmware from source or used this: https://blog.boochow.com/micropython-firmware-for-mbed-boards ?

aivarannamaa, 2019-10-15:
Related issue: https://forum.micropython.org/viewtopic.php?f=12&t=6744

aivarannamaa, 2019-10-15:
I could reproduce this with these firmwares: https://blog.boochow.com/micropython-firmware-for-mbed-boards

As I suspected, it was a buffer issue, but in MicroPython filesystem level, not related to serial connection. I believe you can fix your Thonny before next release by adding the lines shown in the commit.

closed 2019-10-15 by aivarannamaa

aklein9999, 2019-10-16:
Dear Mr. Anamaa,

 

Thank you very much for your help.

 

Currently I am on a business trip, so I couldn't reply earlier.

 

I downloaded the firmware directly from www.microptython.org:

 

http://www.micropython.org/download

 

I can't try your fix before early next week.

 

Should I still have problems I will let you know.

 

Again, thank you very much!

 

Alex Klein

 

Gesendet: Dienstag, 15. Oktober 2019 um 22:35 Uhr

patfrench, 2020-04-16:
Hi, 
I have a same problem with nucleo411RE. I have to install Thonny v3.2.7.
The fix are already ok in this version ?

i have same problem with ESP-WROVER-KITv3 but never problem with ESP32-WROOM.

I don't understand. The firmware for ESP32 come of 
https://micropython.org/download/esp32/
and 
https://micropython.org/download/stm32_nucleo/NUCLEO_F411RE-20200416-v1.12-371-gf534b9976.dfu for my nucleo411RE

```PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "C:\Users\Ppetit\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "C:\Users\Ppetit\AppData\Local\Programs\Thonny\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Ppetit\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in handle_command
    response = handler(cmd)
  File "C:\Users\Ppetit\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 864, in _cmd_upload
    written_bytes = self._upload_file(item["source"], item["target"], notify)
  File "C:\Users\Ppetit\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1461, in _upload_file
    return self._write_file(block_generator(), target, notifier=notifier)
  File "C:\Users\Ppetit\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1094, in _write_file
    result = self._write_file_via_serial(content_blocks, target_path, notifier)
  File "C:\Users\Ppetit\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1166, in _write_file_via_serial
    self._execute_without_output(script)
  File "C:\Users\Ppetit\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 444, in _execute_without_output
    out, err, value = self._execute(script, capture_output=True)
  File "C:\Users\Ppetit\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 438, in _execute
    ok = self._connection.read(2)
  File "C:\Users\Ppetit\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')```

Thank you

aivarannamaa, 2020-04-18:
@patfrench, I didn't understand, which Thonny version, which board and which action gave you this stactrace?

djaychela, 2020-06-16:
I have a similar issue.  I'm using a PyBoard Lite:
`MicroPython v1.12 on 2019-12-20; PYBLITEv1.0 with STM32F411RE`
On Thonny 3.2.7 on macOS 10.14.6.
I can connect without issue to the board and use the REPL, but whenever I execute a command, I get the following:
```Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/workbench.py", line 1598, in event_generate
    handler(event)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/files.py", line 220, in on_toplevel_response
    self.check_update_focus()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/files.py", line 225, in check_update_focus
    self.focus_into(proxy.get_cwd())
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/base_file_browser.py", line 197, in focus_into
    parts = self.split_path(path)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/base_file_browser.py", line 213, in split_path
    return path.split(self.get_dir_separator())
ValueError: empty separator
```

Thonny works without issue with my micro:bit, and mostly works with an M5stickC (with some crashes), but this happens every time I try to use the pyboard.

commit a856b22c (2019-10-15): FIXED #936: Problem uploading files to some STM boards