thonny/thonny #937: need help instalation

gideons1, 2019-10-09:
Hello , Just installed Thonny IDE, seams to wroks... but..
import math or import array   wroks OK 
import machin do not work...the ide do not reconzie the modul
the Thonny was installed by using "c:/...Path..../scripts/pip  install thonnyapp"
it istalled well in 
"C:\Users\user\AppData\Local\Programs\Python\Python37-32\Scripts\thonny.exe -m thonny"
the modul macine files  exisits 
can yuo help
 thkns
Gideon

aivarannamaa, 2019-10-09:
It's really hard to read your message.

How did you install the `machin` (or `macine` or `machine`?) module?

aklein9999, 2019-10-09:
Depending on what hardware you are using some modules are not fully available, e.g. ESP, WiPy, STM and other boards support different underlying modules.
A replacement for machine that often works is pyb. 
So just try: import pyb
or: from pyb import ...

gideons1, 2019-10-09:
thanks
the problem is with only micro-python  modules, they do exit in the correct place...  but not recognized by python interpreter
my son helped  me and found: root cause was missing or corrupted  C++ component used by python to compile new modules
thanks for your quick reply

closed 2019-10-11 by aivarannamaa