thonny/thonny #938: __del__ is not called from shell but from Program. This dousn't happen in IDLE

ulpius117, 2019-10-09:
Take this code-exampla:

lass Test:
    anz=0
    def __init__(self,x):
        self.x=x
        Test.anz+=1
    def __del__(self):
        print("Kill:", self)
        Test.anz-=1
    def __repr__(self):
        return str(self.x)

a=Test(1)
b=Test(2)
c=Test(3)
print(Test.anz)
del b
print(Test.anz)

This gives the correct output::
3
Kill: 2
2

Now try del a from shell:

>>> del a
>>> a
Traceback (most recent call last):
  File "<pyshell>", line 1, in <module>
NameError: name 'a' is not defined
>>> Test.anz
2

del did kill the object a, but didn't call the method __del__, Otherwise it would have written the following text and decreased Test.anz as shown below (which is a screen-copy from IDLE's output):

>>> del a
Kill: 1
>>> Test.anz
1

Is this a bug or please tell me, what I'm doing wrong in Thonny?

ulpius117, 2019-10-09:
Sorry, the Code should look like this:

class Test:
.... anz=0
.... def init(self,x):
........ self.x=x
........ Test.anz+=1
.... def del(self):
........ print("Kill:", self)
........ Test.anz-=1
.... def repr(self):
........ return str(self.x)

closed 2019-10-09 by ulpius117

aivarannamaa, 2019-10-10:
See the note in the gray box: https://docs.python.org/3/reference/datamodel.html#object.__del__

After each command (eg. Run, or executing a single python statement or expression in the shell), Thonny caches reference to each object in global variables table. When you ran the script, it did not know about the existence of object in `b`. When you deleted `a` in the shell, Python could not garbage collect it because Thonny already had refernce to it.

Thonny collects these references in order to support Heap demonstration mode.