thonny/thonny #939: Thonny opens micropython script in one line

stekicar, 2019-10-11:
[image]

I open script from micropython and it does not show correctly. What can be the cause for this?

stekicar, 2019-10-11:
What I have noticed is that if I save file in uPyCraft, Thonny opens same file in one line as shown. Maybe, the problem is with uPyCraft?

aivarannamaa, 2019-10-11:
Thanks for reporting this! I'll try to reproduce this and then I'll can say more

aivarannamaa, 2019-10-11:
I could reproduce this. The problem is indeed with uPyCraft -- at least in Windows it inserts uses only carriage returns for breaking lines. Such line break was used in ancient Macs and Tk-based editor in Thonny does not treat it as a line break.

I'll add a translation from CR to CR+LF or just LF so that people coming from uPyCraft can experience proper linebreaks in Thonny.

commit 94017a43 (2019-10-11): More robust handling of different line endings, fixes #939
closed 2019-10-11 by aivarannamaa